babca/python-gsmmodem #6: Fix unit tests after rapid merging

babca, 2016-09-06:
Some unit tests are now broken.

I suggest to revert a locally cloned repo to the last v0.9 commit 29ca5f5ab030f3907f2d38d60f354a73c8382431, add one commit after the other and reveal which commits break the tests and why.

Anyone able to help?

tomchy, 2017-03-06:
I have just found out that commit 47facc8 breaks "test_decode (test.test_pdu.TestSmsPdu)" unit tests.
Reverting it solves all errors inside test_pdu set.
@babca Could you recheck your commit and update tests if your update was correct?

tomchy, 2017-03-06:
Half of errors inside modem tests are due to the changes made by me inside 4c0187e
It introduces ```AT+CSCS``` command which switches between message encoding schemes, but unit tests and fake modems are not aware of it.
I will try to update them and fix modem SMS tests soon :slightly_smiling_face:

babca, 2017-03-06:
great progress!
https://github.com/babca/python-gsmmodem/commit/47facc8c05b3d4d63c9ff202e5cff033574d2c08 was meant to resolve https://github.com/babca/python-gsmmodem/issues/15
Is test_pdu fixable easily? I will look into it tomorrow. :-)

tomchy, 2017-03-06:
Well - if you roll back your commit it solves all PDU unittests :slightly_smiling_face:

babca, 2017-03-07:
done

tomchy, 2017-03-07:
Could you fix ``coverage`` indicator inside  README.rst ? :slightly_smiling_face:

babca, 2017-03-07:
New indicators added, now in SVG. :slightly_smiling_face:

tomchy, 2017-03-07:
Coveralls still points to the faucamp repository :slightly_smiling_face:
Is this service for free and you can add link to your repo/branch or it is only for premium users? :slightly_smiling_face:

babca, 2017-03-07:
It has the same business model as Github and Travis CI, it's free for open source projects, so I created a new Coveralls account and redirected the badge to it.

babca, 2017-04-27:
thank you tomchy

closed 2017-04-27 by babca