babca/python-gsmmodem #12: sendussd AttributeError: 'NoneType' object has no attribute 'group'

fredyrommy, 2016-09-18:
i was testing the ussd_demo.py and get this error

> Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1097, in __threadedHandleModemNotification
    self._handleUssd(lines)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1408, in _handleUssd
    self._ussdResponse = self._parseCusdResponse(lines)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1441, in _parseCusdResponse
    sessionActive = cusdMatches[0].group(1) == '1'
AttributeError: 'NoneType' object has no attribute 'group'

the Exception just return "None"
sometimes it gives me this error but sometimes it just work perfectly
i'm using raspberry pi 3
ubuntu mate
python 2.7.11
wavecom fastrack m1306b

jthurner, 2016-09-28:
The current code doesn't properly implement USSD replys - if there is an error (unsopported operation/network timeout), the ussd response will only contain a reply code but no text which causes the regex to not match anything, resulting in the error above.

Some starting points for a patch in this issue:
https://github.com/faucamp/python-gsmmodem/issues/43

(stumbled upon this fork by accident, but am not actively using the library anymore so won't be of much further help, sorry)

babca, 2016-09-28:
Hi jthurner, it seems that I missed your `USSD-statuscode-fixes` branch in your repo. Is is safe to merge your [commit](https://github.com/faucamp/python-gsmmodem/commit/46654b97d6495dae343b8bfce7a016d58d1e9cb1)?

fredyrommy, 2016-09-28:
hi @jthurner and @babca what is the best way to fix this problem?
should i use @jthurner's fork?

babca, 2016-09-28:
I merged his code into my `USSD-statuscode-fixes` branch. Please test this: https://github.com/babca/python-gsmmodem/tree/USSD-statuscode-fixes

fredyrommy, 2016-09-29:
doesn't seem to work
first try it doesn't show response properly

> Initializing modem...
> Sending USSD string: *888#
> USSD reply received:
> +CGREG: 2
> USSD session was ended by

second try i got error
> Initializing modem...
> Process Process-2:
> Traceback (most recent cal
>   File "C:\Python27\lib\mu
>     self.run()
>   File "C:\Python27\lib\mu
>     self._target(*self._ar
>   File "D:\Work Data\Serve
>     modem.waitForNetworkCo
>   File "build\bdist.win32\
> erage
>     raise TimeoutException
> TimeoutException: None

third try
> Initializing modem..
> Sending USSD string:
> None

fourth try the device just become unusable like some other program is using it

> could not open port u'com3': WindowsError(5, 'Access is denied.')

babca, 2016-09-29:
Well I'm not on PC at the moment to test it myself, it would be needed to setup a logger and see what is going on. You can setup the logger to stdout/stderr, or to a file (set the log level to debug and show the live log with a command `$ tail -f /tmp/path/to/my/logfile.log`) and see an actual communication with the modem.

I will try to help, but I have two weeks of vacation now. :-)

fredyrommy, 2016-09-29:
here is the AT command log when the error occur

[http://pastebin.com/mSQH7rZe](url)

the response is

`DEBUG: notification: ['+CUSD: 4']`

meanwhile the non-error response is

`DEBUG: notification: ['+CUSD: 1,"Bal: Rp 2000.0\n1. Info pulsa/internet hemat\n2. Bonus Pulsa\n3. Isi Ulang Promo\n4. Cek Kuota\n5. Cek Pulsa Internet\n6. Yuk Solat\n7. Mau Cepat Kaya?\n8. Facebook",15']`

fredyrommy, 2016-09-29:
ups sorry misclicked the comment button

jthurner, 2016-10-06:
> Initializing modem...
> Sending USSD string: *888#
> USSD reply received:
> +CGREG: 2
> USSD session was ended by

The regex in the patch is rubbish, then. I'll fix this, though it may take another weak or so because I'm rather tied up with work atm.

fredyrommy, 2016-10-06:
for now all i can do is this

```
#sessionActive = cusdMatches[0].group(1) == '1'
sessionActive = False
#message = cusdMatches[0].group(2)
message = lines
```
just so it doesn't crash and stop the script but still return the response