babca/python-gsmmodem #13: sms encoding error when trying to send sms

fredyrommy, 2016-10-16:
> DEBUG: write: ATZ
DEBUG: response: ['ATZ\r', 'OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: write: AT+CLAC
DEBUG: response: ['ERROR']
DEBUG: write: AT+WIND?
DEBUG: response: ['+WIND: 50', 'OK']
INFO: Loading Wavecom call state update table
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+6289644000001",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+6289644000001",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: (("SM","BM","SR"),("SM"))', 'OK']
DEBUG: write: AT+CPMS="SM","SM"
DEBUG: response: ['+CPMS: 36,50,36,50', 'OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['ERROR']
DEBUG: write: AT+CLAC
DEBUG: response: ['ERROR']
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "sendsms.py", line 32, in modemp
    sms = modem.sendSms("0895357614989", "test aja", waitForDeliveryReport=True, deliveryTimeout=60)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 829, in sendSms
    self.smsEncoding = 'GSM'
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 677, in smsEncoding
    if not '+CSCS' in self._commands:
TypeError: argument of type 'NoneType' is not iterable

tried AT+CSCS=? on gsmterm the return is

> +CSCS: ("GSM","PCCP437","CUSTOM","HEX")
OK

i'm using raspberry pi 3
ubuntu mate
python 2.7.11
wavecom fastrack m1306b

fredyrommy, 2016-10-27:
anyone can help?

lucasea777, 2016-10-28:
Hello @fredyrommy , this is what I would try:

1. Try using python3
2. Try using this [repo](https://github.com/faucamp/python-gsmmodem) which is more stable:
3. Replace this [line](https://github.com/babca/python-gsmmodem/blob/master/gsmmodem/modem.py#L569
) with `return ['+CSCS']`

By the way, what driver are you using for your modem? (I could not find a driver for a raspberry pi to use my modem)

fredyrommy, 2016-10-29:
hi @lucasea777 
i actually have made some change to made the sms possible, this is what i do
i changed this [line](https://github.com/babca/python-gsmmodem/blob/master/gsmmodem/modem.py#L677) to

>         if self._commands == None:
>             return False

i don't know why my modem will return ['ERROR'] on AT+CLAC
that's what trigger the

>         except CommandError:
>             return None

can you tell me the difference between this 2 repos?
i thought the faucamp's repo has been abandoned

i don't use any driver on my raspberry pi the modem just work i guess linux come with driver for my modem

babca, 2016-10-29:
faucamp's repo is abandoned from 2014, my repo includes all 2014-2016 fixes from all existing forks. Merging all fixes at once could introduce a new set of bugs, so feel free to also try the faucamp's repo if you encounter any problem with my fork, but keep in mind that faucamp's repo doesn't work with python3.

Since your modem doesn't support CLAC command for listing all supported commands, the self.write() on [line 559](https://github.com/babca/python-gsmmodem/blob/master/gsmmodem/modem.py#L559) should raise a CommandError exception and it should switch to an interactive command recognition: [modem.py#L571-L593](https://github.com/babca/python-gsmmodem/blob/master/gsmmodem/modem.py#L571-L593). But it seems it is broken.

Could you try to add `raise CommandError` between lines 559 and 560 to raise the exception manually?
```
...
try:
            # AT+CLAC responses differ between modems. Most respond with +CLAC: and then a comma-separated list of commands
            # while others simply return each command on a new line, with no +CLAC: prefix
            response = self.write('AT+CLAC', timeout=10) # i think this command should raise CommandError, but maybe "ERROR" string recognition is currently broken.
            raise CommandError # for test purpose, try to raise CommandError manually
            if len(response) == 2: # Single-line response, comma separated
                commands = response[0]
                ...
```

fredyrommy, 2016-10-29:
seems like nothing happen
```
DEBUG: write: AT+CLAC
DEBUG: response: ['ERROR']
DEBUG: write: AT+CMGS=27
DEBUG: response: ['> ']
DEBUG: write: 0021000E912698357516949800000ED0A413C40EB7C3A079981D4603
DEBUG: response: ['+CMGS: 32', 'OK']
```

babca, 2016-10-29:
"seems like nothing happen" - we are trying to remove your original exception `TypeError: argument of type 'NoneType' is not iterable`. Is it removed by adding that line into original code?

fredyrommy, 2016-10-29:
no it still give the same error

```
DEBUG: write: AT+CLAC
DEBUG: response: ['ERROR']
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "send.py", line 394, in smsp
    sms = modem.sendSms(DestinationNumber, Text, waitForDeliveryReport=False, deliveryTimeout=10)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 833, in sendSms
    self.smsEncoding = 'GSM'
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 678, in smsEncoding
    if not '+CSCS' in self._commands:
TypeError: argument of type 'NoneType' is not iterable
```

tomchy, 2017-03-06:
Hi!
There are two bugs that have to be fixed concerning your problem:
 - It looks like receiving ```['ERROR']``` from modem after ```AT+CLAC``` command doesn't switch to the interactive command recognition
 - In case of an empty response (no commands available) ```smsSupportedEncoding``` tries to iterates ```None``` received from ```supportedCommands``` call. There should be check for a ```None``` value, although not having knowledge about available commands is quite risky and it is surprising that gsmmodem code was able to go that far without that list. 

But you definitely  forget about debugging it with raising ```ComandError``` exception (as proposed by @babca), because interactive recognition either raises ```TimeoutException``` or return an empty list (not None, fixed by pending pull request with unit tests).

@fredyrommy Could you recheck your isssue?