babca/python-gsmmodem #14: Missing getter for SIM Own Number

petrleocompel, 2016-10-18:
I miss a property phone_number or number or something like that what will represent number of sim in network. I think it would be usefull.

Lot of modems support command ```AT+CNUM```
if it will return just OK the number is missing in own number phonebook 
so we will have to write it to own number phonebook and then AT+CNUM it should work

Or we can read it from the phonebook
Select ON in PB -> ```AT+CPBS="ON"```
Then read the first index -> ```AT+CPBR=0````

-----
```python
def phone_number(self, default_phone_number= None):
	num = None
	if "AT+CNUM" in supported: # supported check
	    resp = self.write('AT+CNUM')
	    if resp is not "OK": # if is not only ok and number was returned - TODO
	    	num = resp
	else:
		actual_phonebook = self.write('AT+CPBS?')
		if actual_phonebook is not "ON": 
		    self.write('AT+CPBS="ON"')
	    try:
	    	num = self.write("AT+CPBR=0")
	    expect Exception: # TODO Change to specific	
	   		num = None

	if num is None
		if default_phone_number is not None:
	    	self.set_phone_number(default_phone_number)
	    	num = default_phone_number
    	else:
    		raise Exception('Number not found')

	return num

def set_phone_number(self, phone_number):
	actual_phonebook = self.write('AT+CPBS?')
	if actual_phonebook is not "ON": 
	    self.write('AT+CPBS="ON"')
    self.write('AT+CPBW=1,"' + phone_number + '"')  # TODO raise exceptions or something like that
```

babca, 2016-10-20:
Zdarec, are you able to fork the repo and do a pull request, or do you need a help?

babca, 2016-10-25:
Yes I am :-)

babca, 2016-10-29:
I've implemented your feature request, fell free to test it. c7a4636e5e20a077e36505158735a66cac50c36f

fredyrommy, 2016-11-19:
this won't work for modem that doesn't support AT+CLAC just like my problem this one
[https://github.com/babca/python-gsmmodem/issues/13](sms encoding error when trying to send sms)

i think this code should be fixed first before adding more function that use AT+CLAC
```
# Get list of supported commands from modem
commands = self.supportedCommands
self._commands = commands
```

```
    @property
    def supportedCommands(self):
        """ :return: list of AT commands supported by this modem (without the AT prefix). Returns None if not known """
        try:
            # AT+CLAC responses differ between modems. Most respond with +CLAC: and then a comma-separated list of commands
            # while others simply return each command on a new line, with no +CLAC: prefix
            response = self.write('AT+CLAC', timeout=10)
            if len(response) == 2: # Single-line response, comma separated
                commands = response[0]
                if commands.startswith(b'+CLAC'):
                    commands = commands[6:] # remove the +CLAC: prefix before splitting
                return commands.split(',')
            elif len(response) > 2: # Multi-line response
                return [cmd.strip() for cmd in response[:-1]]
            else:
                self.log.debug('Unhandled +CLAC response: {0}'.format(response))
                return None
        except (TimeoutException, CommandError):
            # Try interactive command recognition
            commands = []
            checkable_commands = ['^CVOICE', '+VTS', '^DTMF', '^USSDMODE', '+WIND', '+ZPAS', '+CSCS', '+CNUM']

            # Check if modem is still alive
            try:
                response = self.write('AT')
            except:
                raise TimeoutException

            # Check all commands that will by considered
            for command in checkable_commands:
                try:
                    # Compose AT command that will read values under specified function
                    at_command='AT'+command+'=?'
                    response = self.write(at_command)
                    # If there are values inside response - add command to the list
                    commands.append(command)
                except:
                    continue

            # Return found commands
            return commands
```

babca, 2016-11-20:
@fredyrommy are you able to do a pull request? 😃

fredyrommy, 2016-11-20:
@babca no i don't know how

petrleocompel, 2017-01-15:
It seems there left some errors - in detection if command "+CNUM" (missing b - b"+CNUM") is available
and also in parsing - int does not have split function

petrleocompel, 2017-01-15:
I will make pull request

babca, 2017-03-07:
Added in https://github.com/babca/python-gsmmodem/commit/c7a4636e5e20a077e36505158735a66cac50c36f

closed 2017-03-07 by babca