babca/python-gsmmodem #15: encoding error on handling status report

fredyrommy, 2016-10-29:
```
DEBUG: write: AT+CMGS=28
DEBUG: response: ['> ']
DEBUG: write: 0021000E91269835751694980000105479D83D0FAFE769D0BCBE9E97E7
DEBUG: response: ['+CMGS: 35', 'OK']
DEBUG: notification: ['+CDSI: "SR",19']
DEBUG: SMS status report received
DEBUG: write: AT+CPMS="SR"
DEBUG: response: ['+CPMS: 19,50,50,50', 'OK']
DEBUG: write: AT+CMGR=19
DEBUG: response: ['+CMGR: 1,,26', '0006230E9126983575169498610103409544C26101034095448200', 'OK']
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1106, in __threadedHandleModemNotification
    self._handleSmsStatusReport(line)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1274, in _handleSmsStatusReport
    report = self.readStoredSms(msgIndex, msgMemory)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1360, in readStoredSms
    smsDict = decodeSmsPdu(pdu)
  File "build/bdist.linux-armv7l/egg/gsmmodem/pdu.py", line 412, in decodeSmsPdu
    result['time'] = _decodeTimestamp(pduIter)
  File "build/bdist.linux-armv7l/egg/gsmmodem/pdu.py", line 498, in _decodeTimestamp
    return datetime.strptime(dateStr[:-2], '%y%m%d%H%M%S').replace(tzinfo=SmsPduTzInfo(timeZoneStr))
  File "build/bdist.linux-armv7l/egg/gsmmodem/pdu.py", line 60, in __init__
    self._setPduOffsetStr(pduOffsetStr)
  File "build/bdist.linux-armv7l/egg/gsmmodem/pdu.py", line 66, in _setPduOffsetStr
    self._offset = timedelta(minutes=(int(pduOffsetStr) * 15))
ValueError: invalid literal for int() with base 10: '2c'
```

babca, 2016-11-24:
probably fixed in the latest commit, please test.

tomchy, 2017-03-07:
I was able to reproduce your issue inside unittest (based on provided responses) and finally resolve this problem :slightly_smiling_face: 
For some reason negative tie zones were calculated properly, but code for positive ones was different.

I think this issue may be closed after merging PR :slightly_smiling_face:.

babca, 2017-03-07:
Maybe the negative sign is placed in the wrong nibble by the modem?
0x2C -> 0010 1100 -> remove sign from the second nibble -0010 0100 = 0x24 negative * 0.25 = -9h timezone.
https://en.wikipedia.org/wiki/Binary-coded_decimal#Signed_variations

Maybe this bit just should be ignored, then it would mean +9h.

@fredyrommy please let us know your timezone :-) Is it -9 or +9?

tomchy, 2017-03-07:
Currently time zone value is shifted back:
 ``0xC2 --> 0x2C --> 0xC2 --> 0x42``
And afterwards:
``0x42 --> 42 * 15 / 60 --> -10h``

Edit:
I just found that it was too late and I made a mistake:
```python
tzHexVal = int((tzHexVal & 0x0F) * 0x10) + int((tzHexVal & 0x0F) / 0x10)
```
instead of:
```python
tzHexVal = int((tzHexVal & 0x0F) * 0x10) + int((tzHexVal & 0xF0) / 0x10)
```
I will fix that (and add check for parsed value inside unit test) as soon as @fredyrommy will confirm time zone. :slightly_smiling_face:

tomchy, 2017-03-07:
According to the country code used inside your data (+62) it should be Indonesia, which has three time zones: +7, +8 and +9. Additionally, one of timestamps is decoded correctly by an old code (2016-10-30 04:59:44+07:00), so I assume it should be close to +7.

Values 0x2C and 0x28 have small difference, so I thought that someone forgot about BCD-check:
```
0x2C + 0x06 = 0x32 --> 32 * 15min / 1hr = +8
```
@fredyrommy Is it correct? 
This hot fix could be found in a8b449d.

fredyrommy, 2017-03-21:
yes my timezone is UTC +7 Jakarta, Indonesia

babca, 2017-04-27:
@fredyrommy is your issue now resolved? (see #42; download v0.12 release or clone repo)

PR #36 (2017-03-07): PDU: Time zone fix
Fix PDU time zone parsing. Possibly fixes #15, which is reproduced inside new unittest (included).
commit e1852344 (2017-03-07): PDU: fix invalid decoding of large time zone values. Resolves #15.