babca/python-gsmmodem #21: modem disconnecting at AT+CLAC

fredyrommy, 2016-11-19:
when trying to connect to modem i got this error

> DEBUG: write: AT
> DEBUG: response: ['AT\r', 'OK']
> DEBUG: write: ATZ
> DEBUG: response: ['ATZ\r', 'OK']
> DEBUG: write: ATE0
> DEBUG: response: ['ATE0\r', 'OK']
> DEBUG: write: AT+CFUN?
> DEBUG: response: ['+CFUN: 1', 'OK']
> DEBUG: write: AT+CMEE=1
> DEBUG: response: ['OK']
> DEBUG: write: AT+CPIN?
> DEBUG: response: ['+CPIN: READY', 'OK']
> DEBUG: write: AT+CLAC
> DEBUG: write: AT
> Traceback (most recent call last):
>   File "scanmodem.py", line 95, in <module>
>     modem.connect("1234", waitingForModemToStartInSeconds=10)
>   File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 237, in connect
>   File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 577, in supportedCommands
> gsmmodem.exceptions.TimeoutException: None

tried to connect the modem manually from terminal and run AT+CLAC 
it return the supported command but instantly disconnected from computer and then automatically connecting again
i'm using ZTE Vodafone K3806-Z
i just bought this modem because my previous modem Wavecom Fastrack M1306B doesn't support AT+CLAC either

i know this is the problem with the modems but is there any way to bypass this AT+CLAC
i can't afford another modem :(
i mean all the use of this is just checking if AT+CSCS is supported

```python
        if self._commands == None:
            self._commands = self.supportedCommands

        if '+CSCS' in self._commands:
            response = self.write('AT+CSCS?')
```

babca, 2016-11-24:
It sounds to me your modem has a firmware bug and is resetting itself. Have you tried to update or downgrade the firmware? Have you tried to experiment with different speeds (9600, 115200..) and did you try to connect to another channels? Not every channel has full AT command set. You may see the modem as multiple devices such as `/dev/ttyUSB0`, `/dev/ttyUSB1` and `/dev/ttyUSB2`

It is definitely possible to bypass CLAC call by the library. Please connect to your modem manually from terminal and post here your output of AT+CLAC.

fredyrommy, 2016-11-24:
yes i tried all baudrate on both ports
that vodafone modem using 2 ports /dev/ttyACM0 and /dev/ttyACM0 idk why but both just the same disconnect on AT+CLAC

he is the at+clac lists for Vodafone ZTE K3806-Z
```python
['+CGMI', '+CGMM', '+CGMR', '+GMI', '+GMM', '+GMR', 'D', 'DT', 'DP', 'A', 'DL', 'D>', 'H', '+CHUP', 'O', '+CMOD', '+CIND', '+CMEC', '+CMER', '+CACM', '+CAMM', '+CCLK', '+CEER', '+CGSN', '+CIMI', '+CKPD', '+CLAC', '+CNUM', '+CPAS', '+CPBF', '+CPBR', '+CPBS', '+CPBW', '+CPUC', '+GCAP', '+GSN', 'I', 'P', 'T', '+CGACT', '+CGANS', '+CGATT', '+CGCLASS', '+CGCMOD', '+CGDATA', '+CGDCONT', '+CGDSCONT', '+CGEREP', '+CGEQMIN', '+CGEQNEG', '+CGEQREQ', '+CGPADDR', '+CGQMIN', '+CGQREQ', '+CGREG', '+CGSMS', '+CGTFT', '+COPS', '+CPOL', '%ICOPN', '+COPN', '+CSQ', '+CREG', '+CPLS', '+CMGD', '+CMGR', '+CSCA', '+CMGS', '+CMGC', '+CMSS', '+CMGL', '+CMGW', '+CPMS', '+CNMA', '+CMMS', '+CSCB', '&C', '&D', '&F', '&V', '&W', '+CBST', '+CCWE', '+CGAUTO', '+CHSN', '+CMEE', '+CMGF', '+CNMI', '+CR', '+CRC', '+CRES', '+CRLP', '+CSAS', '+CSCS', '+CSNS', '+CSDH', '+CSTA', '+CSMP', '+CSMS', '+CTZR', '+CTZU', '+CV120', '+FCLASS', '+ILRR', 'E', 'L', 'M', 'Q', 'S0', 'S10', 'S2', 'S3', 'S4', 'S5', 'S6', 'S7', 'S8', 'V', 'X', 'Z', '+CFUN', '+CMAR', '+CPIN', '+CSIM', '+CRSM', '+CCHO', '+CCHC', '+CGLA', '+CRLA', '+CUAD', '+PACSP', 'H', '+CAOC', '+CCFC', '+CCWA', '+CCUG', '+CDIP', '+CHLD', '+CLCK', '+CLCC', '+CLIP', '+CLIR', '+COLP', '+CPWD', '+CSSN', '+CUSD', '+VCID', '*TSTCR', '*TSTEV', '*TSTGC', '*TSTMODE', '*TSTMS', '*TSTPD', '*TSTRT', '*TS']
```

tomchy, 2017-03-06:
If it disconnects on ```AT+CLAC``` you may force gsmmodem to use interactive command recognition.
The quickest method to test it is to put:
```python
raise TimeoutException
```
At the beginning of a try section inside modem.py, method ```supportedCommands(self)``` (line 548).