babca/python-gsmmodem #25: The example in ussd is not working.

mwaaas, 2016-11-26:
This the error am gettings.

/Users/francis/PycharmProjects/python-gsmmodem/.venv/bin/python /Users/francis/PycharmProjects/python-gsmmodem/examples/ussd_demo.py
Initializing modem...
Sending USSD string: *140*5*0702729654#
Traceback (most recent call last):
  File "/Users/francis/PycharmProjects/python-gsmmodem/examples/ussd_demo.py", line 41, in <module>
    main()
  File "/Users/francis/PycharmProjects/python-gsmmodem/examples/ussd_demo.py", line 30, in main
    response = modem.sendUssd(USSD_STRING) # response type: gsmmodem.modem.Ussd
  File "/Users/francis/PycharmProjects/python-gsmmodem/gsmmodem/modem.py", line 946, in sendUssd
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None

menon92, 2017-05-10:
I have same problem. Do you get the solution of this problem

mwaaas, 2017-05-18:
Yeah I managed, will find the solution I used then share it with you

menon92, 2017-05-18:
could you please help me now. How can I solve this problem ? I am completely stuck with this problem @babca

mwaaas, 2017-05-19:
Here is the code that I used

#!/usr/bin/env python

"""\
Demo: Simple USSD example

Simple demo app that initiates a USSD session, reads the string response and closes the session
(if it wasn't closed by the network)

Note: for this to work, a valid USSD string for your network must be used.
"""

from __future__ import print_function

import logging

PORT = '/dev/tty.HUAWEIMobile-Pcui'
BAUDRATE = 115200
USSD_STRING = '*140*5*0703488092#'
PIN = None # SIM card PIN (if any)

from gsmmodem.modem import GsmModem

def main():
    print('Initializing modem...')
    #logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
    modem = GsmModem(PORT, BAUDRATE)
    modem.connect(PIN)
    modem.waitForNetworkCoverage(10)
    print('Sending USSD string: {0}'.format(USSD_STRING))
    response = modem.sendUssd(USSD_STRING, 30) # response type:
    # gsmmodem.modem.Ussd
    print('USSD reply received: {0}'.format(response.message))
    if response.sessionActive:
        print('Closing USSD session.')
        # At this point, you could also reply to the USSD message by using response.reply()
        response.cancel()
    else:
        print('USSD session was ended by network.')
    modem.close()

if __name__ == '__main__':
    main()

mwaaas, 2017-05-19:
use this python-gsmmodem-new

pip install python-gsmmodem-new

menon92, 2017-05-20:
Thanks for your reply.  I try your code. In your code `from future import print_function` give me Import Error. for this I change your code like this. and this give me the same error. could you please help me.
```
#!/usr/bin/env python

"""
Demo: Simple USSD example

Simple demo app that initiates a USSD session, reads the string response and closes the session
(if it wasn't closed by the network)

Note: for this to work, a valid USSD string for your network must be used.
"""

#from future import print_function

import logging

#PORT = '/dev/tty.HUAWEIMobile-Pcui'
PORT = '/dev/ttyUSB0'
BAUDRATE = 115200
#USSD_STRING = '1405*0703488092#'
USSD_STRING = 'AA988C3602' # *124# = AA988C3602
PIN = None # SIM card PIN (if any)

from gsmmodem.modem import GsmModem

def main():
	print('Initializing modem...')
	#logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
	modem = GsmModem(PORT, BAUDRATE)
	modem.connect(PIN)
	modem.waitForNetworkCoverage(10)
	print('Sending USSD string: {0}'.format(USSD_STRING))
	response = modem.sendUssd(USSD_STRING, 30) # response type:
	# gsmmodem.modem.Ussd
	print('USSD reply received: {0}'.format(response.message))
	if response.sessionActive:
		print('Closing USSD session.')
		# At this point, you could also reply to the USSD message by using response.reply()
		response.cancel()
	else:
		print('USSD session was ended by network.')
	modem.close()

# if name == 'main':
# 	main()

if __name__ == '__main__':
	main()
```

Thanks @babca

menon92, 2017-05-20:
My modem port:
```
menon@menon-Inspiron-N4050:~$ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 May 20 08:50 /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 1 May 20 08:47 /dev/ttyUSB1
crw-rw---- 1 root dialout 188, 2 May 20 10:16 /dev/ttyUSB2
lrwxrwxrwx 1 root root         7 May 20 08:47 /dev/ttyUSB_utps_diag -> ttyUSB1
lrwxrwxrwx 1 root root         7 May 20 08:47 /dev/ttyUSB_utps_modem -> ttyUSB0
lrwxrwxrwx 1 root root         7 May 20 08:47 /dev/ttyUSB_utps_pcui -> ttyUSB2
menon@menon-Inspiron-N4050:~$ 
```

babca, 2017-05-21:
@menon92 set your PORT to `dev/ttyUSB2`.
The TimeoutException is expected in this case. You can't catch incoming calls, USSDs and SMS messages with `dev/ttyUSB0` port on Huawei modems, because this channel doesn't contain modem notifications. 

If it does not help you, uncomment `logging.basicConfig(...` line and post here your console output.

menon92, 2017-05-21:
**when I set PORT to** `/dev/ttyUSB2` **then output is :**

```
Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB2 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CLAC:&C,&D,&E,&F,&S,&V,&W,E,I,L,M,Q,V,X,Z,T,P,\\Q,\\S,\\V,%V,D,A,H,O,S0,S2,S3,S4,S5,S6,S7,S8,S9,S10,S11,S30,S103,S104,+FCLASS,+ICF,+IFC,+IPR,+GMI,+GMM,+GMR,+GCAP,+GSN,+DR,+DS,+WS46,+CLAC,+CCLK,+CBST,+CRLP,+CV120,+CHSN,+CSSN,+CREG,+CGREG,+CFUN,+GCAP,+CSCS,+CSTA,+CR,+CEER,+CRC,+CMEE,+CGDCONT,+CGDSCONT,+CGTFT,+C', 'ERROR']
DEBUG: write: AT
DEBUG: response: ['OK']
DEBUG: write: AT^CVOICE=?
DEBUG: write: AT+VTS=?
DEBUG: write: AT^DTMF=?
DEBUG: write: AT^USSDMODE=?
DEBUG: write: AT+WIND=?
DEBUG: write: AT+ZPAS=?
DEBUG: write: AT+CSCS=?
DEBUG: write: AT+CNUM=?
DEBUG: write: AT+CGMI
Traceback (most recent call last):
  File "pygsm_modem_send_ussd.py", line 47, in <module>
    main()
  File "pygsm_modem_send_ussd.py", line 29, in main
    modem.connect(PIN)
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/modem.py", line 282, in connect
    if b'simcom' in self.manufacturer.lower() : #simcom modems support DTMF and don't support AT+CLAC
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/modem.py", line 524, in manufacturer
    return self.write('AT+CGMI')[0]
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/modem.py", line 470, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/serial_comms.py", line 133, in write
    self.serial.write(data)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 528, in write
    raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open
```

**And when I set PORT to** `/dev/ttyUSB0` **then output is :**
```
Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CLAC:&C,&D,&E,&F,&S,&V,&W,E,I,L,M,Q,V,X,Z,T,P,\\Q,\\S,\\V,%V,D,A,H,O,S0,S2,S3,S4,S5,S6,S7,S8,S9,S10,S11,S30,S103,S104,+FCLASS,+ICF,+IFC,+IPR,+GMI,+GMM,+GMR,+GCAP,+GSN,+DR,+DS,+WS46,+CLAC,+CCLK,+CBST,+CRLP,+CV120,+CHSN,+CSSN,+CREG,+CGREG,+CFUN,+GCAP,+CSCS,+CSTA,+CR,+CEER,+CRC,+CMEE,+CGDCONT,+CGDSCONT,+CGTFT,+CGEQREQ,+CGEQMIN,+CGQREQ,+CGQMIN,+CGEQNEG,+CGEREP,+CGPADDR,+CGCLASS,+CGSMS,+CSMS,+CMGF,+CSAS,+CRES,+CSCA,+CSMP,+CSDH,+CSCB,+FDD,+FAR,+FCL,+FIT,+ES,+ESA,+CMOD,+CVHU,+CGDATA,+CSQ,+CBC,+CPAS,+CPIN,+CMEC,+CKPD,+CIND,+CMER,+CGATT,+CGACT,+CGCMOD,+CPBS,+CPBR,+CPBF,+CPBW,+CPMS,+CNMI,+CMGL,+CMGR,+CMGS,+CMSS,+CMGW,+CMGD,+CMGC,+CNMA,+CMMS,+FTS,+FRS,+FTH,+FRH,+FTM,+FRM,+CHUP,+CCFC,+CCUG,+COPS,+CLCK,+CPWD,+CUSD,+CAOC,+CACM,+CAMM,+CPUC,+CCWA,+CHLD,+CIMI,+CGMI,+CGMM,+CGMR,+CGSN,+CNUM,+CSIM,+CRSM,+CCLK,+CLVL,+CMUT,+CLCC,+COPN,+CPOL,+CPLS,+CTZR,+CTZU,+CLAC,+CLIP,+COLP,+CDIP,+CTFR,+CLIR,$QCSIMSTAT,$QCCNMI,$QCCLR,$QCDMG,$QCDMR,$QCDNSP,$QCDNSS,$QCTER,$QCSLOT,$QCPINSTAT,$QCPDPP,$QCPDPLT,$QCPWRDN,$QCDGEN,$BREW,$QCSYSMODE,^STSF,^STGI,^STGR,^IMSICHG,^USSDMODE,^SYSINFO,^SYSCFG,^SYSCONFIG,^HS,^DTMF,^CPBR,^CPBW,^HWVER,^HVER,^DSFLOWCLR,^DSFLOWQRY,^DSFLOWRPT,^SPN,^PORTSEL,^CPIN,^SN,^EARST,^CARDLOCK,^CARDUNLOCK,^ATRECORD,^CDUR,^BOOT,^FHVER,^CURC,^FREQLOCK,^FREQPREF,^HSDPA,^GLASTERR,^CARDMODE,^U2DIAG,^PAD,^RSTRIGGER,^PCSCINFO,^CQI,^GETPORTMODE,^CVOICE,^DDSETEX,^pcmrecord,^CMSR,^CMMT,^CMGI,^PCMBUFFER,^CSNR,^PNN,^OPL,^CPNN,^ADCTEMP,^RDCUST,^SETPID,^DIALMODE,^YJCX,^NDISDUP,^DNSP,^DNSS,^DHCP,^AUTHDATA,^CRPN,^ICCID,^FPLMN,^FPLMNCTRL,^NVMBN,^RXDIV,', 'OK']
DEBUG: write: AT^CVOICE=0
DEBUG: response: ['OK']
DEBUG: write: AT^USSDMODE=0
DEBUG: response: ['OK']
INFO: Loading Huawei call state update table
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+880150159999",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+880150159999",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("ME","MT","SM","SR"),("ME","MT","SM","SR"),("ME","MT","SM","SR")', 'OK']
DEBUG: write: AT+CPMS="ME","ME","ME"
DEBUG: response: ['+CPMS: 0,23,0,23,0,23', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['OK']
DEBUG: write: AT+CREG?
DEBUG: response: ['+CREG: 0,1', 'OK']
DEBUG: write: AT+CSQ
DEBUG: response: ['+CSQ: 6,99', 'OK']
Sending USSD string: AA584D3602
DEBUG: write: AT+CUSD=1,"AA584D3602",15
DEBUG: response: ['OK']
Traceback (most recent call last):
  File "pygsm_modem_send_ussd.py", line 47, in <module>
    main()
  File "pygsm_modem_send_ussd.py", line 32, in main
    response = modem.sendUssd(USSD_STRING, 30) # response type:
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/modem.py", line 958, in sendUssd
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None
```

Thanks.

babca, 2017-05-21:
`serial.serialutil.SerialException: Attempting to use a port that is not open`
It seems your modem unexpectedly restarts itself? (Maybe a bug in the modem firmware?)
It fails at simple CLAC command, the response is cut after ~300 characters.

Could you start gsmterm from tools/ folder?
`python gsmterm.py -r /dev/ttyUSB2`
type AT+CLAC, Enter
is the command working properly?
(Press CTRL+] to exit.)

menon92, 2017-05-21:
I run `python gsmterm.py -r /dev/ttyUSB2` and it give me output:
```
Raw terminal connected to /dev/ttyUSB2 at 115200bps.
Press CTRL+] to exit.

ERROR
AT+CLAC
ERROR
AT+CLAC
+CLAC:&C,&D,&E,&F,&S,&V,&W,E,I,L,M,Q,V,X,Z,T,P,\Q,\S,\V,%V,D,A,H,O,S0,S2,S3,S4,S5,S6,S7,S8,S9,S10,S11,S30,S103,S104,+FCLASS,+ICF,+IFC,+IPR,+GMI,+GMM,+GMR,+GCAP,+GSN,+DR,+DS,+WS46,+CLAC,+CCLK,+CBST,+CRLP,+CV120,+CHSN,+CS
+CPMS: 0,50,0,50,0,50
OK
AT+CLAC
Done.
```
I think AT+CLAC command sometimes works and sometimes not working. 

If there is problem with `modem firmware` or `modem unexpectedly restarts` In that case what can i do ? should i try a new modem with the same code ? or anything else i can do . 

Thanks @babca

babca, 2017-05-22:
This seems to be related to the modem not to `python-gsmmodem` library.
Please try it again and type `ATZ` command before `AT+CLAC` command.
Both should output OK on the last line, every time.
I don't think it will help, but you can also try to lower baudrate:
`python gsmterm.py -r /dev/ttyUSB2 -b 9600`

Which modem do you use?

Try a different modem if you can. Huawei E372 and E160 are working fine with `ussd_demo.py`.