babca/python-gsmmodem #33: SMS-DELIVER `from` number decoding error

9dogs, 2017-03-06:
Hey, I've encountered strange bug while decoding some pdu's. I got `EncodingError` exception on some messages while the others are decoding normally.  It seems that a problem is in a function `nibble2octet`:
```python
def nibble2octet(o):
    if o % 2:
        return o / 2 + 1
    else:
        return o / 2
```
In Python 3 this function can produce float values. The value is used later in `unpackSeptets` and comparison with int breaks desired behavior. If it is simple python2->python3 migration issue, it can be fixed simply by `return int(o / 2) + 1`. Can somebody confirm that fix doesn't break anything else?

tomchy, 2017-03-06:
Back in the faucamp/python-gsmmodem repository instead of ```nibble2octet(o)``` there were two pieces of code doing this operation:
```python
addressLen = int(math.ceil(addressLen / 2.0))
```
and:
```python
if addressLen % 2:
    addressLen = int(addressLen / 2) + 1
else:
    addressLen = int(addressLen / 2)
```
There are no more occurrences of ```nibble2octet```, so I think that confirms that this fix doesn't break anything else :slightly_smiling_face:

9dogs, 2017-03-06:
Thank you! I'm making pull request then.

lcnittl, 2017-03-06:
Just a question: Shouldn't `addressLen = int((addressLen + 1) / 2)` give the same values, just without the need of an `if` statement?

babca, 2017-03-06:
The https://github.com/faucamp/python-gsmmodem/commit/df20a9a2960883fbb454bc73c8c5a13f044ef8d4 fix from 2014 was originally part of the codebase until I merged @paolo-losi branch.

The nibble2octet() function was introduced by https://github.com/paolo-losi/python-gsmmodem/commit/0076b9d0ce5eef82af525d48b0f6533467ed18d0 to fix something. I think he wanted to use the same piece of code on both places and he forgot to add int() into his new function.

babca, 2017-03-06:
quick test, both blocks give the same values:
0
1
1
2
2
3
3

```
import math

def testblock1(addressLen):
	if addressLen % 2:
	    addressLen = int(addressLen / 2) + 1
	else:
	    addressLen = int(addressLen / 2)
	print (addressLen)

testblock1(0)
testblock1(1)
testblock1(2)
testblock1(3)
testblock1(4)
testblock1(5)
testblock1(6)

def testblock2(addressLen):
	addressLen = int(math.ceil(addressLen / 2.0))
	print (addressLen)

testblock2(0)
testblock2(1)
testblock2(2)
testblock2(3)
testblock2(4)
testblock2(5)
testblock2(6)
```

babca, 2017-03-06:
I suggest to change `nibble2octet()` function to:

```
def nibble2octet(o):
    addressLen = int((addressLen + 1) / 2)
```

closed 2017-03-06 by babca