babca/python-gsmmodem #41: +CFUN 1,0

kardasjan, 2017-03-19:
It is not true that every modem responds with single int value in CFUN command.

modem.py +223
`cfun = int(lineStartingWith('+CFUN:', self.write('AT+CFUN?'))[7:]) # example response: +CFUN: 1`

The line could be replaced with following one if you don't need to use the command.
`cfun = lineStartingWith('+CFUN:', self.write('AT+CFUN?'))[7:] # example response: +CFUN: 1`

I don't see that as a real issue, though mentioning it seemed responsible.

brianbienvenu, 2018-10-27:
Hit the same issue with a U-Blox L210, and modified that section on modem.py#222 to this:

        try:
            cfun = lineStartingWith('+CFUN:', self.write('AT+CFUN?'))[7:] # example response: +CFUN: 1,0
            if not cfun.startswith('1'):
                self.write('AT+CFUN=1')
    ...

tomchy, 2018-11-06:
Thank you for pointing that out!
After a quick research I found out that gsmmodem behaviour can be confirmed by GSM 07.07 specification and numerous other AT commad reference manuals.

Could you provide an explanation/spec for the alternative return values?
I think the idea of this project is to make a flexible and adaptive python library, so a small update should be more than welcome :slightly_smiling_face:

babca, 2018-11-13:
Is there any way to detect this modem from available AT commands? It could be added to detection algorithm (modem.py L238+), which could then activate modem-specific features.

lcnittl, 2021-11-02:
Should be fixed with #86

closed 2021-11-02 by lcnittl