babca/python-gsmmodem #52: Old version while installing from pip

tarelda, 2017-09-11:
Dunno why, just installs 0.11 instead of 0.12...

babca, 2017-09-11:
Hi @tarelda,
thanks for pointing that out, that is interesting. I've registered v0.10, v0.11, v0.12 in pip repository: https://pypi.python.org/pypi/python-gsmmodem-new
but only v0.11 is available from command line... 

`pip3 install python-gsmmodem-new==999999`
 Could not find a version that satisfies the requirement python-gsmmodem-new==999999 (from versions: 0.11)
No matching distribution found for python-gsmmodem-new==999999

I don't know the reason ATM.

tarelda, 2017-09-12:
@babca, I am not familiar with uploading to pypi, but look at this:
`root@0209c693656a:/# pip3 install --verbose --no-cache-dir -I python-gsmmodem-new
Converted retries value: Retry(total=5, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=5, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
Converted retries value: Retry(total=5, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=5, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
Collecting python-gsmmodem-new
  1 location(s) to search for versions of python-gsmmodem-new:
  * https://pypi.python.org/simple/python-gsmmodem-new/
  Getting page https://pypi.python.org/simple/python-gsmmodem-new/
  Starting new HTTPS connection (1): pypi.python.org
  "GET /simple/python-gsmmodem-new/ HTTP/1.1" 200 242
  Analyzing links from page https://pypi.python.org/simple/python-gsmmodem-new/
    Found link https://pypi.python.org/packages/56/77/f9ffcd4930c1e0ad33bac4208cd76abfeb1ae4af0acd35fd5751dbe4380a/python-gsmmodem-new-0.11.tar.gz#md5=c9d0f278ad3c3ccf343ee6d3b18dd3fb (from https://pypi.python.org/simple/python-gsmmodem-new/), version: 0.11
  Using version 0.11 (newest of versions: 0.11)
  "GET /packages/56/77/f9ffcd4930c1e0ad33bac4208cd76abfeb1ae4af0acd35fd5751dbe4380a/python-gsmmodem-new-0.11.tar.gz HTTP/1.1" 200 96092
  Downloading python-gsmmodem-new-0.11.tar.gz (96kB)
  Downloading from URL https://pypi.python.org/packages/56/77/f9ffcd4930c1e0ad33bac4208cd76abfeb1ae4af0acd35fd5751dbe4380a/python-gsmmodem-new-0.11.tar.gz#md5=c9d0f278ad3c3ccf343ee6d3b18dd3fb (from https://pypi.python.org/simple/python-gsmmodem-new/)
    100% |████████████████████████████████| 102kB 713kB/s
  Running setup.py (path:/tmp/pip-build-em6q9d34/python-gsmmodem-new/setup.py) egg_info for package python-gsmmodem-new
    Running command python setup.py egg_info
    running egg_info
    creating pip-egg-info/python_gsmmodem_new.egg-info
    writing pip-egg-info/python_gsmmodem_new.egg-info/PKG-INFO
    writing requirements to pip-egg-info/python_gsmmodem_new.egg-info/requires.txt
    writing dependency_links to pip-egg-info/python_gsmmodem_new.egg-info/dependency_links.txt
    writing top-level names to pip-egg-info/python_gsmmodem_new.egg-info/top_level.txt
    writing manifest file 'pip-egg-info/python_gsmmodem_new.egg-info/SOURCES.txt'
    warning: manifest_maker: standard file '-c' not found

    reading manifest file 'pip-egg-info/python_gsmmodem_new.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'pip-egg-info/python_gsmmodem_new.egg-info/SOURCES.txt'
  Source in /tmp/pip-build-em6q9d34/python-gsmmodem-new has version 0.11, which satisfies requirement python-gsmmodem-new from https://pypi.python.org/packages/56/77/f9ffcd4930c1e0ad33bac4208cd76abfeb1ae4af0acd35fd5751dbe4380a/python-gsmmodem-new-0.11.tar.gz#md5=c9d0f278ad3c3ccf343ee6d3b18dd3fb
Collecting pyserial>=3.1.1 (from python-gsmmodem-new)
  1 location(s) to search for versions of pyserial:
  * https://pypi.python.org/simple/pyserial/
  Getting page https://pypi.python.org/simple/pyserial/
  "GET /simple/pyserial/ HTTP/1.1" 200 2102
  Analyzing links from page https://pypi.python.org/simple/pyserial/
    Skipping link https://pypi.python.org/packages/0c/16/2c0b371fc3387945304ee2e0db24cd80d7ef8df1c4fe85617d8cccd8fc06/pyserial-2.3-py2.5.egg#md5=e2944f767a14a91058e67f73526b7f2b (from https://pypi.python.org/simple/pyserial/); unsupported archive format: .egg
    Found link https://pypi.python.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl#md5=0e555d61700e0b95a15d8162092c5299 (from https://pypi.python.org/simple/pyserial/), version: 3.4
    Skipping link https://pypi.python.org/packages/1c/ba/5c96d18a04787a7cd90ef353ad8f0a20012bd2fd2f43eb62196e15c46b54/pyserial-2.7.win32_py3k.exe#md5=c6fb580ae7763671297794b8a1d91c9e (from https://pypi.python.org/simple/pyserial/); unsupported archive format: .exe
    Found link https://pypi.python.org/packages/1f/3b/ee6f354bcb1e28a7cd735be98f39ecf80554948284b41e9f7965951befa6/pyserial-3.2.1.tar.gz#md5=7142a421c8b35d2dac6c47c254db023d (from https://pypi.python.org/simple/pyserial/), version: 3.2.1
    Found link https://pypi.python.org/packages/22/64/b76b13dae3ec81ee2e6424c6d30b01e335a7e5d33c5c0014cac0d3e117e4/pyserial-3.2.tar.gz#md5=51aed83f45d94e070de2ff706cffa2b4 (from https://pypi.python.org/simple/pyserial/), version: 3.2
    Skipping link https://pypi.python.org/packages/24/0c/1caef2e25d2c44780ead837a0d68c942c77ec92ddccaafe5333293868e08/pyserial-3.0.win32.exe#md5=a23d97fee6f816a17f546c02265cca43 (from https://pypi.python.org/simple/pyserial/); unsupported archive format: .exe
    Found link https://pypi.python.org/packages/2c/c2/2a391a69730c05a646f27e71e965cf8a0e9ccd72c0f0b870b74d47f3c32b/pyserial-3.2-py2.py3-none-any.whl#md5=bc25ea197f9ec3e99360aed38e149fd4 (from https://pypi.python.org/simple/pyserial/), version: 3.2
    Found link https://pypi.python.org/packages/3c/d8/a9fa247ca60b02b3bebbd61766b4f321393b57b13c53b18f6f62cf172c08/pyserial-3.1.1.tar.gz#md5=2f72100de3e410b36d575e12e82e9d27 (from https://pypi.python.org/simple/pyserial/), version: 3.1.1
    Found link https://pypi.python.org/packages/42/68/41bab8a5e3e50705f25358f043181242fa197f313aa0476908d9852c8dc4/pyserial-3.0.1.tar.gz#md5=c8521e49f8852196aac39431e0901703 (from https://pypi.python.org/simple/pyserial/), version: 3. [... cut ...]
Converted retries value: Retry(total=0, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=0, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
Converted retries value: Retry(total=0, connect=None, read=None, redirect=None) -> Retry(total=Retry(total=0, connect=None, read=None, redirect=None), connect=None, read=None, redirect=None)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.`
It seems that pip is trying to download it from https://pypi.python.org/simple/python-gsmmodem-new/ and indeed 0.12 is missing here

lcnittl, 2021-11-03:
Should be fixed with newest release

closed 2021-11-03 by lcnittl