babca/python-gsmmodem #58: Failed to receive sms from example

M010, 2018-01-25:
I am trying to receive sms, using code from example.
Program connected to  gsmmodem. 
I receive my sms once, when I plug in new gsm modem the first time.
What could be the problem?

babca, 2018-11-06:
sms_handler_demo.py has stdout logging enabled by default. Please post full log, we couldn't help without it.

EricCBonet, 2019-03-19:
Hi again, 

When I'm using my laptop this software is working excellent. On a raspberry Pi 3B+, with raspbian lite and python3.6 compiled from source, sending messages works but receiving them does not. I cannot tell from the error messages what is causing this issue. 

```
pi@raspberrypi:~/python-gsmmodem/examples $ python3 sms_handler_demo.py 
Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB2 at 96000bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CLAC:&C,&D,&F,&V,E,I,L,M,Q,V,X,Z,T,P,D,A,H,O,S0,S2,S3,S4,S5,S6,S7,S8,S9,S10,S11,S30,S103,S104,+FCLASS,+ICF,+IFC,+IPR,+GMI,+GMM,+GMR,+GCAP,+GSN,+DR,+DS,+WS46,+CLAC,+CCLK,+CBST,+CRLP,+CV120,+CHSN,+CSSN,+CREG,+CGREG,+CFUN,+GCAP,+CSCS,+CSTA,+CR,+CEER,+CRC,+CMEE,+CGDCONT,+CGDSCONT,+CGTFT,+CGEQREQ,+CGEQMIN,+CGQREQ,+CGQMIN,+CGEQNEG,+CGEREP,+CGPADDR,+CGCLASS,+CGSMS,+CSMS,+CMGF,+CSAS,+CRES,+CSCA,+CSMP,+CSDH,+CSCB,+FDD,+FAR,+FCL,+FIT,+ES,+ESA,+CMOD,+CVHU,+CGDATA,+CSQ,+CBC,+CPAS,+CPIN,+CMEC,+CKPD,+CIND,+CMER,+CGATT,+CGACT,+CGCMOD,+CPBS,+CPBR,+CPBF,+CPBW,+CPMS,+CNMI,+CMGL,+CMGR,+CMGS,+CMSS,+CMGW,+CMGD,+CMGC,+CNMA,+CMMS,+FTS,+FRS,+FTH,+FRH,+FTM,+FRM,+CHUP,+CCFC,+CCUG,+COPS,+CLCK,+CPWD,+CUSD,+CAOC,+CACM,+CAMM,+CPUC,+CCWA,+CHLD,+CIMI,+CGMI,+CGMM,+CGMR,+CGSN,+CNUM,+CSIM,+CRSM,+CCLK,+CLVL,+CMUT,+CLCC,+COPN,+CPOL,+CPLS,+CTZR,+CTZU,+CLAC,+CLIP,+COLP,+CDIP,+CTFR,+CLIR,$QCSIMSTAT,$QCCNMI,$QCCLR,$QCDMG,$QCDMR,$QCDNSP,$QCDNSS,$QCTER,$QCSLOT,$QCPINSTAT,$QCPDPP,$QCPDPLT,$QCPWRDN,$QCDGEN,$BREW,$QCSYSMODE,$QCCTM,^IMSICHG,^STSF,^STGI,^STGR,^CELLMODE,^SYSINFO,^DIALMODE,^SYSCFG,^SYSCONFIG,^HS,^DTMF,^CPBR,^CPBW,^HWVER,^HVER,^DSFLOWCLR,^DSFLOWQRY,^DSFLOWRPT,^SPN,^PORTSEL,^CPIN,^SN,^EARST,^CARDLOCK,^CARDUNLOCK,^ATRECORD,^CDUR,^BOOT,^FHVER,^CURC,^FREQLOCK,^FREQPREF,^HSPA,^HSUPA,^HSDPA,^GLASTERR,^CARDMODE,^U2DIAG,^SETPID,^SCSITIMEOUT,^CQI,^GETPORTMODE,^CVOICE,^DDSETEX,^pcmrecord,^CMSR,^CMMT,^CMGI,^RDCUST,^FPLMN,^FPLMNCTRL,^USSDMODE,^YJCX,^NCELL,', 'OK']
DEBUG: write: AT^CVOICE=0
DEBUG: response: ['ERROR']
DEBUG: write: AT^USSDMODE=0
DEBUG: response: ['OK']
INFO: Loading Huawei call state update table
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+31640191919",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+31640191919",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("ME","MT","SM","SR"),("ME","MT","SM","SR"),("ME","MT","SM","SR")', 'OK']
DEBUG: write: AT+CPMS="ME","ME","ME"
DEBUG: response: ['+CPMS: 15,23,15,23,15,23', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['OK']
Waiting for SMS message...
pi@raspberrypi:~/python-gsmmodem/examples $ 
```

devegied, 2019-04-14:
I was also playing on Raspberry Pi 3B and found this example not working as expected. I think is the value of timeout in `join` function [call](https://github.com/babca/python-gsmmodem/blob/master/examples/sms_handler_demo.py#L35): `2**31` is `2147483648` and `_thread.TIMEOUT_MAX` is `9223372036.0` on Raspbian Python 3.5. So this `2**31` should be acceptable value on Pi but for some reason this call to `join` returns immediately.
Maybe more portable way would be to call `join` without arguments and it will wait _thread.TIMEOUT_MAX seconds (292 years on Raspberry Pi or 50 days on 32bit Windows)
I created PR #77 - maybe someone can test it on another platform (I can test only on Raspberry Pi)

kikil592, 2020-06-23:
> Hi again,
>
> When I'm using my laptop this software is working excellent. On a raspberry Pi 3B+, with raspbian lite and python3.6 compiled from source, sending messages works but receiving them does [...]

Same problem for me on rpi3b and ubuntu mate, the python script works perfectly on w10 however. (it is more than the huge time in the thread to make the listening last) Did you found an answer @EricCBonet ?

I've a E1692 huawei gsm modem

alza54, 2021-12-15:
Same issue for me.
```
Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 19200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT
DEBUG: response: ['OK']
DEBUG: write: AT^CVOICE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+VTS=?
DEBUG: response: ['+VTS: (0-9,*,#,A,B,C,D),,(1-255)', 'OK']
DEBUG: write: AT^DTMF=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT^USSDMODE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+WIND=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+ZPAS=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+CSCS=?
DEBUG: response: ['+CSCS: ("GSM","HEX","IRA","PCCP437","UCS2","8859-1")', 'OK']
DEBUG: write: AT+CNUM=?
DEBUG: response: ['OK']
DEBUG: write: AT+CGMI
DEBUG: response: ['Quectel_Ltd', 'Quectel_M95', 'Revision: MTK 0828', 'OK']
DEBUG: write: AT+CGMI
DEBUG: response: ['Quectel_Ltd', 'Quectel_M95', 'Revision: MTK 0828', 'OK']
DEBUG: write: AT+ZPAS?
DEBUG: response: ['+CME ERROR: 100']
INFO: Unknown/generic modem type - will use polling for call state updates
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+48501200777",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+48501200777",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("SM","ME","MT"),("SM","ME","MT"),("SM","ME","MT")', 'OK']
DEBUG: write: AT+CPMS="ME","ME","ME"
DEBUG: response: ['+CPMS: 10,10,10,10,10,10', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['OK']
Waiting for SMS message...
```

(Raspberry Pi)