babca/python-gsmmodem #59: About handling several consecutive texts/sms?

XGBXGB, 2018-01-31:
I'm currently testing by sending several texts to the gsm modem (at most 7 at a time), and I think it ruins the threads resulting in TimeoutExceptions. Anyone also experiencing this/tried this?

Sample error:
`File "C:\Users\***\Anaconda3\lib\site-packages\gsmmodem\modem.py", line 1354, in _handleSmsReceived
    self.smsReceivedCallback(sms)
  File "main.py", line 24, in handleSms
    sms.reply("Reply: "+sms.text)
  File "C:\Users\***\Anaconda3\lib\site-packages\gsmmodem\modem.py", line 66, in reply
    return self._gsmModem.sendSms(self.number, message)
  File "C:\Users\***\Anaconda3\lib\site-packages\gsmmodem\modem.py", line 902, in sendSms
    result = lineStartingWith('+CMGS:', self.write(text, timeout=35, writeTerm=CTRLZ))
  File "C:\Users\***\Anaconda3\lib\site-packages\gsmmodem\modem.py", line 458, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "C:\Users\***\Anaconda3\lib\site-packages\gsmmodem\serial_comms.py", line 144, in write
    raise TimeoutException()
`

tomchy, 2018-02-05:
What do you mean by sending several messages at once? You would like to create several connections to a single modem and send them simultaneously or just send a very long text, that consists of 7 parts?

I tested this library once (around a year ago) by sending a single, extremely long message (something like 1000+ characters) and it worked quite well. If that is your issue I may rerun this test :slightly_smiling_face:

If not, please remember that normally you have just a single serial interface, that should be blocked by a single process. Couple of modems actually register itself with multiple virtual serial ports, but their functionality usually changes among them.

Also a code snipped may give us a clue what you are trying to achieve and will allow us to reproduce this error.

XGBXGB, 2018-03-03:
I apologize for not being able to reply right away. What's happening was I was only using a single modem, which is both the receiver and the sender, when i send multiple text like spamming it, it tends to stop replying to the 5th text, but when I included another modem, one modem is the receiver (converts the texts into textfiles) and one modem is the sender (waits and reads textfiles), the issue doesn't happen anymore.

Now I'm going to try using one modem as the receiver and multiple modems as the sender. Is this possible?

tomchy, 2018-04-17:
I think it should be possible. Unfortunately I still have not checked this module with multiple modems, but that case definitely should be considered if one day I get some time to make a small set of system tests :slightly_smiling_face:

babca, 2018-11-06:
You can always use the modem for single task at a time. Don't make multiple connections to the same modem.

Using multiple modems is fine.

Just run a separate thread using correct serial settings for each modem.
It does not matter whether you run multiple scripts by hand in terminal, or you write a custom initscript/daemon/etc which will create the threads automatically.