babca/python-gsmmodem #60: Concatenated SMS problem when using UCS2 coding

TishSerg, 2018-03-22:
Hello! I trying to send long SMSs. When I send long SMS (eg. consisting of 7 PDUs) using GSM-7 coding it works fine. But when I send long SMS using UCS2 coding (even consisting of 2 PDUs) - on receiving side text is corrupted. It seems like an encoding problem, isn't it?

Source text (I trying to send this):
```Python
File		:	beacon.py - Location reporter service
# Version	:	0.8.2
# Date		:	2018-03-12
# Author	:	TishSerg (Ukraine)
# Email		:	TishSerg@gmail.com
# ==============================================================================

import sys
import time
import threading
import socket
import urllib.request
import subprocess
import re
import beacon
if __name__ == "__main__":
    from location_provider import LocationProvider
    from modem_comm import Modem
else:
    from .location_provider import LocationProvider
    from .modem_comm import Modem

#GPS_PORT = 'COM15'  # DEBUG
#GSM_PORT = 'COM17'  # DEBUG
GPS_PORT = '/dev/ttyS1'  # NanoPi's UART1
GSM_PORT = '/dev/ttyS2'  # NanoPi's UART2
GSM_RTS_PIN = 13  # NanoPi's PA2 pin
GSM_CTS_PIN = 15  # NanoPi's PA3 pin
GSM_STATUS_PIN = 21  # NanoPi's PC1 pin
GSM_PWRKEY_PIN = 23  # NanoPi's PC2 pin (Active High)
```

This text will be encoded using UCS2 due to <TAB> symbols near the beginning. I programmatically limit this text to fit it into 1 then 2 PDUs. Results on the picture:
[image]
First (correctly coded) message - source text was limited to 1 PDU (UCS2).
Second (incorrectly coded) message - same source text was limited to 2 PDUs (UCS2).

The Modem I use is SIM800C.

tomchy, 2018-04-17:
1. How do you switch into UCS2 encoding?
2. It is always worth checking if you are using the correct library (with -new suffix) and the latest release (0.12).

TishSerg, 2018-04-17:
1. Your library switches it automatically when fails to encode text using GSM-7 coding. File `pdu.py`, lines 299-307:
```Python
    # Encode message text and set data coding scheme based on text contents
    try:
        encodedTextLength = len(encodeGsm7(text))
    except ValueError:
        # Cannot encode text using GSM-7; use UCS2 instead
        encodedTextLength = len(text)
        alphabet = 0x08 # UCS2
    else:
        alphabet = 0x00 # GSM-7
```
2. Of course, library version is correct (v0.12 w/ -new suffix)

TishSerg, 2018-04-17:
Sorry. I've misclicked.

tomchy, 2018-04-17:
Could you dump produced PDUs?

After line 919 of modem.py add:

```
self.log.error("Debug PDUs.---\nMessage: {0}\n---\n---PDU: ---\n".format(text))
for p in pdus:
    self.log.error(str(codecs.encode(p.data, 'hex_codec')).upper())
self.log.error("---")
```

TishSerg, 2018-04-17:
Sorry, at this time I don't have that hardware. When I will be able to do this, I'll let you know.

babca, 2018-11-06:
@TishSerg Any update on this? :-)

TishSerg, 2018-11-06:
Oops. I've completely forgotten about it. 
Now, I'm not working with hardware anymore, so cannot reproduce this :( 
Sorry.

babca, 2018-11-06:
Anyone up and running any modem with this library? Could you try to send the text/code from the first post (above) in SMS?

@TishSerg it could be a generic issue, let's see. Thanks for the prompt reply. :-)