babca/python-gsmmodem #64: Failed to send SMS with the sample code

y2kbug-hk, 2018-08-28:
May I know what is going wrong?
The SIM card can get a network and send a SMS with a phone.
Thanks.

> $ sudo python3 tools/sendsms.py -i /dev/ttyUSB0 85212345678 test --debug
Connecting to GSM modem on /dev/ttyUSB0...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT
DEBUG: response: ['OK']
DEBUG: write: AT^CVOICE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+VTS=?
DEBUG: response: ['+VTS: (0-9,*,#,A,B,C,D),(1-255)', 'OK']
DEBUG: write: AT^DTMF=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT^USSDMODE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+WIND=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+ZPAS=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+CSCS=?
DEBUG: response: ['+CSCS: ("IRA","GSM","UCS2","HEX","PCCP","PCDN","8859-1")', 'OK']
DEBUG: write: AT+CNUM=?
DEBUG: response: ['OK']
DEBUG: write: AT+CGMI
DEBUG: response: ['SIMCOM_Ltd', 'OK']
DEBUG: write: AT+DDET=1
DEBUG: response: ['OK']
DEBUG: write: AT+CGMI
DEBUG: response: ['SIMCOM_Ltd', 'OK']
DEBUG: write: AT+ZPAS?
DEBUG: response: ['+CME ERROR: 100']
INFO: Unknown/generic modem type - will use polling for call state updates
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+85290100000",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+85290100000",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT")', 'OK']
DEBUG: write: AT+CPMS="ME","ME","ME"
DEBUG: response: ['+CPMS: 0,20,0,20,0,20', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['+CMS ERROR: 325']
DEBUG: write: AT+CNMI=2,1,0,1,0
DEBUG: response: ['OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['OK']
Checking for network coverage...
DEBUG: write: AT+CREG?
DEBUG: response: ['+CREG: 0,1', 'OK']
DEBUG: write: AT+CSQ
DEBUG: response: ['+CSQ: 25,0', 'OK']

Sending SMS message...
DEBUG: write: AT+CSCS=?
DEBUG: response: ['+CSCS: ("IRA","GSM","UCS2","HEX","PCCP","PCDN","8859-1")', 'OK']
DEBUG: write: AT+CSCS="GSM"
DEBUG: response: ['OK']
DEBUG: write: AT+CMGS=17
DEBUG: response: ['> ']
DEBUG: write: 0021000BA15862030969F0000004F4F29C0E
DEBUG: response: ['+CMS ERROR: 38']
Traceback (most recent call last):
  File "tools/sendsms.py", line 116, in <module>
    main()
  File "tools/sendsms.py", line 58, in main
    send_sms(args)
  File "tools/sendsms.py", line 103, in send_sms
    sms = modem.sendSms(args.destination, text, waitForDeliveryReport=args.deliver)
  File "/home/y2kbug/.local/lib/python3.6/site-packages/gsmmodem/modem.py", line 924, in sendSms
    result = lineStartingWith('+CMGS:', self.write(str(pdu), timeout=35, writeTerm=CTRLZ)) # example: +CMGS: xx
  File "/home/y2kbug/.local/lib/python3.6/site-packages/gsmmodem/modem.py", line 486, in write
    raise CmsError(data, int(errorCode))
gsmmodem.exceptions.CmsError: CMS 38

babca, 2018-08-28:
Hello, try to use an international formatted number.

y2kbug-hk, 2018-08-28:
OMG... +85212345678 works now! What a silly mistake... Thank you very much!

closed 2018-08-28 by y2kbug-hk