babca/python-gsmmodem #66: modem.connect() does not work with Python 3

guangyuzhang, 2018-11-06:
Here is my code:
`
from gsmmodem.modem import GsmModem
import logging

if __name__ == '__main__':
    port = '/dev/ttyUSB0'
    baud_rate = 115200
    logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
    modem = GsmModem(port, baud_rate)

    try:
        modem.connect()
        modem.waitForNetworkCoverage(10)
    except Exception as e:
        print('Failed to connect the modem: ' + str(e))
        if modem.alive:
            modem.close()
    else:
        print('Connected!')
        if modem.alive:
            modem.close()
`
Run it with python 3.6:
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
Failed to connect the modem: None

Run it with python 2.7:
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['ATZ\r', 'OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+WIND?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+CGMI
DEBUG: response: ['+CGMI: COMWAY', 'OK']
DEBUG: write: AT+ZPAS?
DEBUG: response: ['+CME ERROR: 100']
INFO: Unknown/generic modem type - will use polling for call state updates
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "8613800100500", 145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("SM", "ME", "SM_P", "ME_P", "MT"), ("SM", "ME", "SM_P", "ME_P", "MT"), ("SM", "ME", "SM_P", "ME_P", "MT")', 'OK']
DEBUG: write: AT+CPMS="ME"
DEBUG: response: ['+CPMS: 0, 100, 3, 50, 3, 50', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['ERROR']
WARNING: Incoming SMS notifications not supported by modem. SMS receiving unavailable.
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['OK']
DEBUG: write: AT+CREG?
DEBUG: response: ['+CREG: 0, 1', 'OK']
INFO: +CREG check disabled due to invalid response or unsupported command
DEBUG: write: AT+CSQ
DEBUG: response: ['+CSQ: 8, 99', 'OK']
Connected!

[test_connect.txt](https://github.com/babca/python-gsmmodem/files/2551983/test_connect.txt)

tomchy, 2018-11-06:
Hi guangyuzhang!
Could you rerun the test and post which line of code causes your issue?

guangyuzhang, 2018-11-07:
> Hi guangyuzhang!
> Could you rerun the test and post which line of code causes your issue?

INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: AT
DEBUG: write: ATZ
Failed to connect the modem: None
^CTraceback (most recent call last):
  File "test_connect.py", line 13, in <module>
    modem.connect()
  File "/usr/local/lib/python3.6/dist-packages/python_gsmmodem_new-0.12-py3.6.egg/gsmmodem/modem.py", line 210, in connect
  File "/usr/local/lib/python3.6/dist-packages/python_gsmmodem_new-0.12-py3.6.egg/gsmmodem/modem.py", line 458, in write
  File "/usr/local/lib/python3.6/dist-packages/python_gsmmodem_new-0.12-py3.6.egg/gsmmodem/serial_comms.py", line 144, in write
gsmmodem.exceptions.TimeoutException: None

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_connect.py", line 18, in <module>
    modem.close()
  File "/usr/local/lib/python3.6/dist-packages/python_gsmmodem_new-0.12-py3.6.egg/gsmmodem/serial_comms.py", line 62, in close
  File "/usr/local/lib/python3.6/dist-packages/pyserial-3.4-py3.6.egg/serial/serialposix.py", line 474, in close
KeyboardInterrupt

guangyuzhang, 2018-11-07:
> Hi guangyuzhang!
> Could you rerun the test and post which line of code causes your issue?

INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: write: AT
DEBUG: response: ['OK']
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['OK']
Failed to connect the modem: 'NoneType' object is not subscriptable

goon94, 2019-08-09:
Hi, i have the same error.
Do you already know how to solve the problem?