babca/python-gsmmodem #69: [new feature]web server

superbool, 2019-01-07:
Hi,do you have a plan to make a simple gsmmodem web server?
I think it's a very good idea .Think about it ,you can send sms from you brower or make phone calls.

babca, 2019-01-07:
People implement this library into their projects directly, however, even basic webserver would be useful. Feel free to make it :-) it could be a separated project, more or less.

superbool, 2019-01-10:
Yes,so I created a repository [flask-gsm](https://github.com/superbool/web-gsm) based on [python-gsmmodem-new](https://github.com/babca/python-gsmmodem) and flask,but it count not read sms. Can you help me?`gsm_number` worked fine. but the sms read callback function never received msg.

EricCBonet, 2019-03-20:
I'm working on this for sms messages. 

Have a local computer (or raspberry Pi..if i can figure out how to receive messages from the pi) which has a GSM dongle. You can then send and receive messages from that dongle from a browser. I have a use case for this project in Lesotho. 

I'm using a lot of the code from the [aileen project](https://github.com/aileenproject/aileen). Right now it's in a super early stage and for convenience during development my passwords are hard coded. Once I get it functioning I'll create a new repo for it on GitHub and would love for people to check it out :)