babca/python-gsmmodem #70: Pass audio recordings to call

BenasPaulikas, 2019-01-20:
Hey is this tool capable of calling a number and playing pre-recorded audio file ?

babca, 2019-03-11:
Hi @BenasPaulikas, it would be nice to be able to build a callcenter-like automated phone system. However, the library currently doesn't provide high-level methods for passing an audio file, only sending (and receiving) DTMF tones is implemented. Adding audio functionality is possible with this library, no-one has implemented it yet though.

BenasPaulikas, 2019-03-11:
Yeah no worries. I ended up using Asterisk and it's perfect. Ty

ferfalam, 2022-06-01:
Hello @BenasPaulikas 

I want to know how you solved this problem.

Can you help ?

Thanks.

BenasPaulikas, 2022-06-02:
@ferfalam 

https://www.spinics.net/lists/pjsip/msg20671.html

ferfalam, 2022-06-02:
Thank you @BenasPaulikas .

But Is it possible to use python-gsmmodem to that directly ?

BenasPaulikas, 2022-06-03:
@ferfalam that I don't know.