babca/python-gsmmodem #80: Unable to read old messages

stevecohenfr, 2019-08-06:
Hi,

I try to read SMS received while python-gsmmodem was not running but not able to make it work...

    #!/usr/bin/env python
    
    from __future__ import print_function
    from datetime import datetime
    from gsmmodem.modem import GsmModem, Sms
    
    import logging
    
    PORT = '/dev/ttyUSB0'
    BAUDRATE = 115200
    PIN = None  # SIM card PIN (if any)
    
    
    def handleSms(sms):
        print(u'== SMS message received ==\nFrom: {0}\nTime: {1}\nMessage:\n{2}\n'.format(sms.number, sms.time, sms.text))
    
    
    def main():
        print('Initializing modem...')
        # Uncomment the following line to see what the modem is doing:
        logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
        modem = GsmModem(PORT, BAUDRATE, smsReceivedCallbackFunc=handleSms)
        modem.smsTextMode = False
        modem.connect(PIN)
        #modem.processStoredSms(False)
        smss = modem.listStoredSms(status=Sms.STATUS_ALL, delete=True);
        print(str(len(smss)) + ' messages found')
        for i in range(len(smss)):
            sms = smss[i]
            handleSms(sms)
        print('Waiting for new SMS message...')    
        try:    
            modem.rxThread.join(2**31) # Specify a (huge) timeout so that it essentially blocks indefinitely, but still receives CTRL+C interrupt signal
        finally:
            modem.close();
    
    if __name__ == '__main__':
        main()

Here are the logs :

    Initializing modem...
    INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
    DEBUG: write: ATZ
    DEBUG: response: [u'OK']
    DEBUG: write: ATE0
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CFUN?
    DEBUG: response: [u'+CFUN: 1', u'OK']
    DEBUG: write: AT+CMEE=1
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CPIN?
    DEBUG: response: [u'+CPIN: READY', u'OK']
    DEBUG: write: AT+CLAC
    DEBUG: write: AT
    DEBUG: response: [u'OK']
    DEBUG: write: AT^CVOICE=?
    DEBUG: response: [u'^CVOICE: (0)', u'OK']
    DEBUG: write: AT+VTS=?
    DEBUG: response: [u'+VTS: (0-9,A-D,*,#)', u'OK']
    DEBUG: write: AT^DTMF=?
    DEBUG: response: [u'ERROR']
    DEBUG: write: AT^USSDMODE=?
    DEBUG: response: [u'^USSDMODE: (0-1)', u'OK']
    DEBUG: write: AT+WIND=?
    DEBUG: response: [u'ERROR']
    DEBUG: write: AT+ZPAS=?
    DEBUG: response: [u'ERROR']
    DEBUG: write: AT+CSCS=?
    DEBUG: response: [u'+CSCS: ("IRA","UCS2","GSM")', u'OK']
    DEBUG: write: AT+CNUM=?
    DEBUG: response: [u'OK']
    DEBUG: write: AT^CVOICE=0
    DEBUG: response: [u'ERROR']
    DEBUG: write: AT^USSDMODE=0
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CGMI
    DEBUG: response: [u'huawei', u'OK']
    DEBUG: write: AT+CGMI
    DEBUG: response: [u'huawei', u'OK']
    INFO: Loading Huawei call state update table
    DEBUG: write: AT+COPS=3,0
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CMGF=0
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CSCA?
    DEBUG: response: [u'+CSCA: "002B00330033003600390035003000300030003600390035",145', u'OK']
    DEBUG: write: AT+CSMP=49,167,0,0
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CSCA?
    DEBUG: response: [u'+CSCA: "002B00330033003600390035003000300030003600390035",145', u'OK']
    DEBUG: write: AT+CPMS=?
    DEBUG: response: [u'+CPMS: ("SM","ME"),("SM","ME"),("SM","ME")', u'OK']
    DEBUG: write: AT+CPMS="ME","ME","ME"
    DEBUG: response: [u'+CPMS: 0,20,0,20,0,20', u'OK']
    DEBUG: write: AT+CNMI=2,1,0,2
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CLIP=1
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CRC=1
    DEBUG: response: [u'OK']
    DEBUG: write: AT+CVHU=0
    DEBUG: response: [u'ERROR']
    DEBUG: write: AT+CMGL=4
    DEBUG: response: [u'OK']
    0 messages found
    Waiting for new SMS message...

I only receive message when python-gsmmodem is running and I'm unable to read messages that I received before...
I also tried processStoredSms() but nothing append.

Thanks

tomchy, 2021-02-13:
Hi @stevecohenfr !
Could you connect to the modem and send the following queries?
`AT+CNMI=?`
`AT+CNMI?`

It is possible that you have to use a custom setting, by providing the `AT_CNMI` modem argument.

stevecohenfr, 2021-02-13:
> Hi @stevecohenfr !
> Could you connect to the modem and send the following queries?
> `AT+CNMI=?`
> `AT+CNMI?`
> 
> It is possible that you have to use a custom setting, by providing the `AT_CNMI` modem argument.

Hello @tomchy ,

thanks got your answer. I fixed my issue long time ago 🙂
Here is my script to read short, long and in memory messages : https://gist.github.com/stevecohenfr/8d3908da4ed39169992e407261f4c0e6

tomchy, 2021-02-13:
Hello @stevecohenfr !
Thank you for the script! I see that it has addressed two issues, that were reported: fixing the sendsms script and reading multipart messages.

So, what was the issue? Just the lack of SMS handler or was it somewhere deeper in the modem code?

stevecohenfr, 2021-02-14:
Yes, just a lake in the exemple code.
For multipart sms my script just keep all part un an array, sort it and concatenate all parts.

For the send message issue it was not a real issue. You cannot use the dongle if it used by another script. So you cannot call a script to send your message while another script is waiting for incoming messages. So I added Pyro to allow the "sending script" to communicate with "reading script" and ask it to send the message

lcnittl, 2021-11-03:
Closing as resolved. Feel free to reopen if this is not the case.

closed 2021-11-03 by lcnittl