babca/python-gsmmodem #83: ValueError: invalid literal for int() with base 10: '250,0'

Roundik, 2020-01-12:
Hello, 

Unfortunately, I can’t tell the modem yet.
One thing I know is chinese on 16 ports.

```
def main():
    d=0
    p=24
    max_ports=30
    imsin = ''
    path = Path(__file__).parent.absolute()
    while d < max_try:
        while p < max_ports:
            print('try ',p, ' :port; ')
            PORT = 'COM'+str(p)
            logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
            modem = GsmModem(PORT, BAUDRATE, smsReceivedCallbackFunc=handleSms)
            modem.connect(PIN)
            modem.rxThread.join(2**8)
            modem.close();
            p += 1
        d += 1
```

```
try  24  :port;
INFO: Connecting to modem on port COM24 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT
DEBUG: response: ['OK']
DEBUG: write: AT^CVOICE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+VTS=?
DEBUG: response: ['+VTS: (0-9,*,#,A,B,C,D),,(1-255)', 'OK']
DEBUG: write: AT^DTMF=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT^USSDMODE=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+WIND=?
DEBUG: response: ['+WIND: (250,257)[,(0,1)]', 'OK']
DEBUG: write: AT+ZPAS=?
DEBUG: response: ['+CME ERROR: 100']
DEBUG: write: AT+CSCS=?
DEBUG: response: ['+CSCS: ("GSM","HEX","IRA","PCCP437","UCS2","8859-1")', 'OK']
DEBUG: write: AT+CNUM=?
DEBUG: response: ['OK']
DEBUG: write: AT+WIND?
DEBUG: response: ['+WIND: 250,0', 'OK']
Traceback (most recent call last):
  File "D:\yandex/sms_receive.py", line 75, in <module>
    main()
  File "D:\yandex/sms_receive.py", line 56, in main
    modem.connect(PIN)
  File "D:\yandex\Python38\lib\site-packages\gsmmodem\modem.py", line 268, in connect
    if int(wind[7:]) != 50:
ValueError: invalid literal for int() with base 10: '250,0'
```

Thanks

furryDev1lOwO, 2022-11-10:
This is freaking damn bug with Wind implementation

furryDev1lOwO, 2022-11-10:
This is a poor implementation of the WIND protocol for some of the older modems
All you need to do is to remove the block of code related to WIND in the modem.py library
I'll be working on a commit the next few days.