babca/python-gsmmodem #92: Still Alive?

lcnittl, 2020-11-19:
Hi @babca,

First off, thanks for maintaining this repository!
Given the amount of unopened yet old issues and pull requests, I was wondering if you still find the time to continue this work?

In case not, would you mind to get some help? I would not really like to create yet another new python-gsmmodem package on PyPI (given that yours is already the "new" one).

Happy to hear your thoughts,
lcnittl

tomchy, 2021-02-12:
Hi @lcnittl!
I do agree with you that creating a third repository will make a mess in pypi. Do you have some new features that you would like to merge - or - just have some resources to handle tickets/PRs?

I do not know if it is actively maintained, but for sure it is used in production.

I hope that @babca will join this conversation. He did a great job on updating and fixing the library. Since there are so many modems out there, having a community seems to be a reasonable choice to keep it in a good shape.

lcnittl, 2021-02-13:
Hi @tomchy,

@babca has reached out to me after opening this issue. The quint essence of my reply was:

> I wanted (maybe best after considering the PRs) do some rewrite of the package to make it fitter for python3 and also trim the setup.py to a minimum (cf [DMFO](https://github.com/lcnittl/DMFO)). If it is okay for you, I would also switch to use `setuptools_scm` for python version control, which allows to have no hard-coded version in any file, but just read it from the Git version tags. This works quite nicely (cf above link, or [sphinx-contrib.email](https://github.com/sphinx-contrib/email) or [sphinx-contrib.scm](https://github.com/sphinx-contrib/scm) or [pyPFC](https://github.com/lcnittl/pyPFC)).

I have not received an answer to this email – I hope it did no end up in babca's spam.

sim0nx, 2021-09-07:
I was wondering the same thing ... any updates on this :-) ?

lcnittl, 2021-09-07:
Not really, unfortunately :-/

lcnittl, 2021-11-02:
@babca has added me to the repo 🙇

closed 2021-11-02 by lcnittl