babca/python-gsmmodem #93: Question: support for windows?

rtrahms, 2021-03-06:
New to using GSM modems, and have used them in raspbian/linux contexts, but unclear on whether this would work in windows (and how to modify the '/dev/ttyS0' identifier to a windows equivalent).  Any suggestions/advice would be appreciated - thanks.

lcnittl, 2021-03-07:
Without having tried, from a `pyserial` perspective, using `"COM1"` (or whatever COM-port is actually in action) instead of `"/dev/ttyS0"` should work on Windows. Other than that, I do not think that here are linux-only methods in use.  As I am at the moment not able to test it on windows in the next days, I would be curious about your result.

rtrahms, 2021-03-14:
It seems to have issues, but I am not sure if it is Windows related.  For my test, I am using a USB GSM modem with a SIM800C chip.  The device shows up in Windows with a standard USB device on a COM port that I have assigned to COM10.  A serial connection to the device using PuTTY works just fine, I am able to issue AT commands with appropriate responses.  No PIN is required to connect.

I have also been able to talk to the device using PySerial commands (simple class code below):

import serial
import time

import serial.tools.list_ports as port_list

class GsmBasic:
    def __init__(self, port, baud):
        self.port = port
        self.baud = baud
        self.isOpen = False

        self.gsm = serial.Serial()
        self.gsm.port = port
        self.gsm.baudrate = baud

    def connect(self):
        self.gsm.open()

        self.isOpen = self.gsm.isOpen()

        if self.isOpen == False:
            print("error opening gsm!")

        return

    def disconnect(self):
        self.gsm.close()
        self.isOpen = False

    def reset(self):

        if self.isOpen == False:
            print("error, modem not active")
            return

        command = "ATZ\n"
        self.gsm.write(command.encode())

        # assume echo
        self.gsm.readline()  # assume echo for command
        print(command)
        response = self.gsm.readline()
        print(response.decode("Ascii"))

    def get_model_info(self):

        if self.isOpen == False:
            print("error, modem not active")
            return

        command = "AT+GSV\n"
        self.gsm.write(command.encode())

        # assume echo
        self.gsm.readline()  # assume echo for command
        print(command)
        response = self.gsm.readline()
        print(response.decode("Ascii"))
        response = self.gsm.readline()
        print(response.decode("Ascii"))
        response = self.gsm.readline()
        print(response.decode("Ascii"))

    def check_sig_strength(self):

        if self.isOpen == False:
            print("error, modem not active")
            return

        command = "AT+CSQ\n"
        self.gsm.write(command.encode())

        # assume echo
        self.gsm.readline()  # assume echo for command
        print(command)
        response = self.gsm.readline()
        print(response.decode("Ascii"))

The following sequence times out, not sure why:

gsmModem = GsmModem("COM10",115200) 
gsmModem.connect()

tomchy, 2021-03-16:
Hi @rtrahms !
Your issue is quite similar to that one: https://github.com/babca/python-gsmmodem/issues/94

I still cannot figure out, what is the difference between sending `ATZ` from gsmmodem and your sample code.

Do you have some tools to check for differences (like USB-UART converters)?
Could you verify that if you send `ATZ\r`, the sample code still works?

lcnittl, 2021-03-17:
> Incidentally, running "python identify-modem.py -b 115200 COM10" yields the same timeout result.

Can you run `python identify-modem.py -d -b 115200 COM10` and post the output? Thanks!

GarrettIRL, 2021-03-17:
@rtrahms I'm having the same issue running on ubuntu so not a windows problem. As far as I can tell the script isn't writing the initial ATZ command until after the timeout exception is raised. Its possibly an issue with the version of python, pyserial we are both using as with this noted behavior the program would never have worked, which seems contrary as people have obviously been using it successfully.

lcnittl, 2021-03-17:
I am running 5c816ba (`pip3 install git+git://github.com/babca/python-gsmmodem@5c816ba`) successfully on Ubuntu 20.04.1 LTS with Python 3.8.5.

lcnittl, 2021-03-17:
I remember that I had problems when installing from PyPI - but I do not recall if it was this timeout behavior, or rather something else. Can you try installing from GitHub (as per my previous comment) and report back if there was a change in behavior?

GarrettIRL, 2021-03-17:
> I am running [5c816ba](https://github.com/babca/python-gsmmodem/commit/5c816bae121c552850e88238da15344fd299bb0e) (`pip3 install git+git://github.com/babca/python-gsmmodem@5c816ba`) successfully on Ubuntu 20.04.1 LTS with Python 3.8.5.

I've just pip3 uninstalled the PyPi install and reinstalled the branch from git as you suggested.

 

> I remember that I had problems when installing from PyPI - but I do not recall if it was this timeout behavior, or rather something else. Can you try installing from GitHub (as per my previous comment) and report back if there was a change in behavior?

Unfortunately the same timeout exception is happening, I've tried on both a laptop running ubuntu and a RPI running raspbian both are having the same issue.

tomchy, 2021-03-17:
Could you dump the list of downloaded packages (`pip3 feeeze`)?

Maybe this is something completely irrelevant, but there is a `serial` as well as `pyserial` package in pip. I remember that only one of them was usable.

GarrettIRL, 2021-03-18:
interesting suggestion I do have both pyserial==3.4 & serial==0.0.97 showing up on the Ubuntu 18.04.5 laptop with PIP3 freeze, but as a bit of a sanity check earlier on I downloaded python-gsmmodem on a RPI and was getting the same error, the PI only has pyserial 3.4 on it, so possibly not the issue. I'm a bit hesitant to remove the serial package (not sure what i downloaded it for or if it's likely to break something else) based on the fact its not on the PI and I'm having the same complications.

lucasmpr, 2022-03-29:
I found a problem that could be causing this.

Line 50, serial_comms.py. The modem is opened with dstcts=true and dsrdtr=true. As I didn't connect dts and rts, the messages were not flowing to the device.

Removing this initialization solved it for me.