babca/python-gsmmodem #94: Timeout exception raised for any command.

GarrettIRL, 2021-03-13:
Hi All

I've looked through the other issues and have seen similar behavior to what I'm getting but none of the fixes seem to work. Is this repo still maintained?  I have a SIMCOMM SIM7600G-H modem from the debug I'm assuming that the response from ATZ is not what the program is expecting, I've tried in putty and ATZ is returning OK which seems like the normal response. What am I missing? I've placed some logger calls around the code to see what is hanging and replaced ATZ with just AT, nothing seems to get me past these few lines of code. Any help would be appreciated. 
Debug info:

Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
INFO: Making the serial connection
INFO: Serial connection sucess
DEBUG: Before command error 1
DEBUG: write: ATZ
DEBUG: In the write function, where timeout is happening
DEBUG: In first else condition
DEBUG: In second else condition
Traceback (most recent call last):
  File "dial_polling_demo.py", line 72, in <module>
    main()
  File "dial_polling_demo.py", line 33, in main
    modem.connect(PIN)
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/modem.py", line 213, in connect
    self.write('ATZ') # reset configuration
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/modem.py", line 468, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/serial_comms.py", line 151, in write
    raise TimeoutException()
lib.exceptions.TimeoutException: None

tomchy, 2021-03-13:
Hi @GarrettIRL !
Do you know which serial connection parameters have worked in putty?
Maybe there is a flow control required, or a different newline sequence (see `TERMINATOR` in `modem.py`)?

GarrettIRL, 2021-03-13:
Hi @tomchy thanks for the quick reply, the same serial port /dev/ttyUSB0 @ 115200 works fine in putty and in fact with stand alone python scripts. Possibly, I would have thought that AT commands for all simcom modules have the same terminator, even the older GSM modules the library was intended for and the newer LTE modems.

GarrettIRL, 2021-03-13:
I have just tried changing TERMINATOR = '\r' on line 20 of modem.py to TERMINATOR = '\n\r' and it didn't work. The documentation for the sim7600G-h would suggest that '\r' is correct see below.
"The "AT" or "at" or "aT" or "At" prefix must be set at the beginning of each Command line. To terminate a
Command line enter <CR>."

lcnittl, 2021-03-13:
May I ask what timeout value you use?

tomchy, 2021-03-13:
Thank you for testing!
Have you tried to use the `waitingForModemToStartInSeconds` argument in the `connect` method?
It simply sends `AT` command every 0.5 second until the other end responds or the timeout is reached.

If that will pass, but it will still fail in the same place, I suggest typing the `ATZ` manually - maybe this command requires a longer time to get a response.

GarrettIRL, 2021-03-13:
> May I ask what timeout value you use?

The time out value in the write function of modem.py is set to 10 which is the default, just incase I've just changed it to 50 and no difference, still experiencing a timeout exception writing the first ATZ.

GarrettIRL, 2021-03-13:
> Thank you for testing!

No problem I really want to get this working so any help is much appreciated.

> Have you tried to use the `waitingForModemToStartInSeconds` argument in the `connect` method?

I've just tried based on your recommendation.
 
> It simply sends `AT` command every 0.5 second until the other end responds or the timeout is reached.
> 
> If that will pass, but it will still fail in the same place, I suggest typing the `ATZ` manually - maybe this command requires a longer time to get a response.

Unfortunately it didn't work either. It just sent a series of AT and got stuck at the same part on each write then eventually times out. I've included the fist 2 AT writes debug below with the error at the end, but the other 8 or so are the same.

Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
INFO: Making the serial connection
INFO: Serial connection success
DEBUG: write: AT
DEBUG: made it back to modem.py in write function
DEBUG: In the write function, where timeout is happening
DEBUG: In the main if statement
DEBUG: In the main if statement after nested if
DEBUG: In main else statement
DEBUG: In second else condition serial_comms line 150
DEBUG: write: AT
DEBUG: made it back to modem.py in write function
DEBUG: In the write function, where timeout is happening
DEBUG: In the main if statement
DEBUG: In the main if statement after nested if
DEBUG: In main else statement
DEBUG: In second else condition serial_comms line 150

Traceback (most recent call last):
  File "dial_polling_demo.py", line 72, in <module>
    main()
  File "dial_polling_demo.py", line 33, in main
    modem.connect(PIN)
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/modem.py", line 213, in connect
    self.write('ATZ') # reset configuration
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/modem.py", line 469, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "/home/gar/Documents/radio_scripts/serial_tests_new/split/python-gsmmodem_basic/lib/serial_comms.py", line 157, in write
    raise TimeoutException()
lib.exceptions.TimeoutException: None

lcnittl, 2021-03-17:
> 
> 
> > May I ask what timeout value you use?
> 
> The time out value in the write function of modem.py is set to 10 which is the default, just incase I've just changed it to 50 and no difference, still experiencing a timeout exception writing the first ATZ.

I see. The reason for my question was that if one uses `2**31` as timeout, like suggested in:

https://github.com/babca/python-gsmmodem/blob/5c816bae121c552850e88238da15344fd299bb0e/examples/incoming_call_demo.py#L52-L55

newer version of python directly continue/raise timeout exception without _any_ delay.

But this might be a different story, as it has to do with threading in this case.

lcnittl, 2021-03-17:
Have you tried running [identify-modem](https://github.com/babca/python-gsmmodem/blob/master/tools/identify-modem.py) with debug mode so far? This should print the responses from the modem. Like this we can double check if responses are matching expected responses.

GarrettIRL, 2021-03-17:
> Have you tried running [identify-modem](https://github.com/babca/python-gsmmodem/blob/master/tools/identify-modem.py) with debug mode so far? This should print the responses from the modem. Like this we can double check if responses are matching expected responses.

I've just tried now but am experiencing the same timeout problem.

Just to get to the bottom of this, I've connected an FTDI cable and ran the program connected up to a RPI also with an FTDI breakout. While the program is running nothing is actually being sent over the line, its only after lib.exceptions.TimeoutException: None happens does the ATZ get written and shows on the serial terminal on the PI. No matter what value of timeout is use in the write function of modem.py the ATZ is simply not getting written out until after the exception is raised.

lcnittl, 2021-03-17:
What is the actual output from `identify-modem.py`? At what point does the timeout exception occur? And which python version are you running it in?

```sh
# identify-modem.py -b 115200 -p 0000 /dev/ttyS0
args: Namespace(baud='115200', debug=False, pin='0000', port='/dev/ttyS0', wait=0)
Connecting to GSM modem on /dev/ttyS0...

== MODEM INFORMATION ==

Manufacturer: SIMCOM_Ltd
Model: SIMCOM_SIM800L
Revision: Revision:1308B07SIM800L16

IMEI: 000000000000000
IMSI: 000000000000000

Network: A1
Signal strength: 28
```
```sh
# identify-modem.py -b 115200 -p 0000 -d /dev/ttyS0
args: Namespace(baud='115200', debug=True, pin='0000', port='/dev/ttyS0', wait=0)
Connecting to GSM modem on /dev/ttyS0...

== MODEM DEBUG INFORMATION ==

ATI ['SIM800 R13.08', 'OK']
AT+CGMI: ['SIMCOM_Ltd', 'OK']
AT+CGMM: ['SIMCOM_SIM800L', 'OK']
AT+CGMR: ['Revision:1308B07SIM800L16', 'OK']
AT+CFUN=?: ['+CFUN: (0,1,4),(1)', 'OK']
AT+WIND=?: ['+CME ERROR: 100']
AT+WIND?: ['+CME ERROR: 100']
AT+CPMS=?: ['+CPMS: ("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT"),("SM","ME","SM_P","ME_P","MT")', 'OK']
AT+CNMI=?: ['+CNMI: (0-3),(0-3),(0,2),(0,1),(0,1)', 'OK']
AT+CVHU=?: ['+CVHU: (0-1)', 'OK']
AT+CSMP?: ['+CSMP: 49,167,0,0', 'OK']
AT+GCAP: ['+GCAP: +CGSM', 'OK']
AT+CPIN? ['+CPIN: READY', 'OK']
AT+CLAC: ['+CME ERROR: 100']
```

GarrettIRL, 2021-03-17:
There is no output from identify-modem.py as the timeout occurs running this too. In each script of the module the .connect function is called, this function attempts to write ATZ to the modem as one of its first steps, on writing the program hangs with the timeout exception (after experimenting with 2 FTDI cables I've learnt that nothing actually got written over the serial connection) then for some unknown reason after the timeout occurs the ATZ command gets written out over serial, but by this time the module is no longer looking for the response as the program has excited. 

I honestly cant see how this is occurring for me while others have successfully managed to get it to work. The only thing I can think about is if the modem is supposed to send something before ever being written to, if so this could initiate the module to exert the correct behavior, but this is unlikely and I have never come across a modem like this. Just for reference I've tried on both SIM7600G-H and SIM800L based modems.

GarrettIRL, 2021-03-18:
"And which python version are you running it in?" 

Sorry just saw the above part of your question both python 3.6.9 and have since removed the package with pip3 uninstall and downloaded it again with pip to use with python 2.7.16. The problem has continued.

lcnittl, 2021-11-02:
Where you ever able to resolve this?

petterhj, 2021-11-08:
I seem to have a similar issue:
* Modem: SIMCOM SIM900A
* Python: 3.9.5

```
identify-modem.py -b 57600 COM6 -d
args: Namespace(port='COM6', baud='57600', pin=None, debug=True, wait=0)
Connecting to GSM modem on COM6...
Traceback (most recent call last):
  File "C:\projects\playground\sim900amini\pygsm\.venvwin\Scripts\identify-modem.py", line 90, in <module>
    main()
  File "C:\projects\playground\sim900amini\pygsm\.venvwin\Scripts\identify-modem.py", line 51, in main
    modem.connect(args.pin, waitingForModemToStartInSeconds=args.wait)
  File "C:\projects\playground\sim900amini\pygsm\.venvwin\lib\site-packages\gsmmodem\modem.py", line 211, in connect
    self.write('ATZ') # reset configuration
  File "C:\projects\playground\sim900amini\pygsm\.venvwin\lib\site-packages\gsmmodem\modem.py", line 464, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "C:\projects\playground\sim900amini\pygsm\.venvwin\lib\site-packages\gsmmodem\serial_comms.py", line 144, in write
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None
```` 
Using Pyserial directly:
```py
import serial

ser = serial.Serial(
    port="COM6",
    baudrate=57600,
    bytesize=serial.EIGHTBITS,
    parity=serial.PARITY_NONE,
    stopbits=serial.STOPBITS_ONE,
)
print(ser.is_open)
ser.write(b"AT\r")
while True:
    line = ser.readline()
    print(line)
```
```
True
b'AT\r\r\n'
b'OK\r\n'
```
Using a terminal (Termite, 8N1, parity=none, flow control=none, append CR+LF, RTS=off, DTR=off):
```
AT
AT

OK
AT+IPR?
AT+IPR?

+IPR: 57600

OK
```

waldner, 2021-11-23:
I'm having exactly the same problem as @GarrettIRL . The modem is a "teleorigin RL900L2.X.X.X.X.X", the stack traces show `ATZ` being written then the timeout exception, here an example:

```
$ ~/.local/bin/identify-modem.py -d -p 0000 -b 115200 /dev/ttyUSB0 
args: Namespace(port='/dev/ttyUSB0', baud='115200', pin='0000', debug=True, wait=0)
Connecting to GSM modem on /dev/ttyUSB0...
Traceback (most recent call last):
  File "/home/waldner/.local/bin/identify-modem.py", line 90, in <module>
    main()
  File "/home/waldner/.local/bin/identify-modem.py", line 51, in main
    modem.connect(args.pin, waitingForModemToStartInSeconds=args.wait)
  File "/home/waldner/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 211, in connect
    self.write('ATZ') # reset configuration
  File "/home/waldner/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 464, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "/home/waldner/.local/lib/python3.9/site-packages/gsmmodem/serial_comms.py", line 144, in write
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None
```

Same thing when I try `modem.connect(None)`  (the sim has no pin).

jaydenpung, 2025-04-24:
For anyone who is able to connect with putty and get response, this would likely be the fix:
https://github.com/babca/python-gsmmodem/pull/118