babca/python-gsmmodem #95: Error: 'GsmModem' object has no attribute 'rxThread'

drejc12, 2021-04-07:
Hi,
when trying out `incoming_call_demo.py` example with the demo python code on Raspberry Pi I get the error when calling this  `modem.rxThread.join(2**31)` and in `modem.close()`:

`AttributeError: 'GsmModem' object has no attribute 'rxThread'`

I believe it's not package `gsmmodem `problem, but what it is, I have no clue - I am just a begginer in python/RPi..

I am making a project which will set one pin on the RPi when certain number call the GSM modem which is plugged in RPi.
Does anyone can help me with an already made code or example for my problem?

Looking forward for your reply and resolving my project issues.

drejc12, 2021-04-07:
I am done with the error I posted. Sorry..

Now I run the same code `incoming_call_demo.py` without error, however, the code runs utill the end. Is there a problem with my GSM modem? 

My modem is Huawei E173 - I am not sure if this one really supports calls?

Is there any other wrong step/thing I am making?

lcnittl, 2021-04-07:
Try to remove the `2**31` and run the script again.

drejc12, 2021-04-07:
Thanks, @lcnittl, your suggestion did work, appreciate that!

Now my script is executing and waiting for an incoming call.

[image]

Now I don't know what to do. I have called the modem number several times, but the ring seems to have a different ringing tone than other "normal" numbers.

Trying to figure out what on my side is wrong now.