babca/python-gsmmodem #102: TimeoutException with Sim800 module

HamidMousavi-git, 2022-04-28:
Hi
I want to test this library with Sim800 module but it raise timeout exception
i test the module with putty and also with below code and every thigs is ok:

`import serial
import time

ser = serial.Serial(port='COM3', baudrate=9600, timeout=3)
ser.write(b"AT\r\n")
time.sleep(1)
ser.write(b'AT+CMGF=1\n\r')
time.sleep(1)
ser.flushInput()
ser.flushOutput()
command = b'AT+CMGL="ALL"\n\r'#gets incoming sms that has not been read -("REC UNREAD","REC READ","STO UNSENT","STO SENT","ALL")
ser.write(command)
time.sleep(1)
print('ser.inWaiting(): {0}'.format(ser.inWaiting()))
print(ser.read_all())
while ser.inWaiting():
    print(ser.read_all())
ser.close()     `

but when i run identify-modem.py or other examples it raise timeout exception

> python.exe c:/Users/Hamid/Desktop/python-gsmmodem-master/tools/identify-modem.py  COM3 -b 9600  -d        
args: Namespace(port='COM3', baud='9600', pin=None, debug=True, wait=0)
Connecting to GSM modem on COM3...
Traceback (most recent call last):
  File "c:\Users\Hamid\Desktop\python-gsmmodem-master\tools\identify-modem.py", line 90, in <module>
    main()
  File "c:\Users\Hamid\Desktop\python-gsmmodem-master\tools\identify-modem.py", line 51, in main
    modem.connect(args.pin, waitingForModemToStartInSeconds=args.wait)
  File "C:\Users\Hamid\AppData\Local\Programs\Python\Python310\lib\site-packages\gsmmodem\modem.py", line 211, in connect
    self.write('ATZ') # reset configuration
  File "C:\Users\Hamid\AppData\Local\Programs\Python\Python310\lib\site-packages\gsmmodem\modem.py", line 464, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "C:\Users\Hamid\AppData\Local\Programs\Python\Python310\lib\site-packages\gsmmodem\serial_comms.py", line 144, in write
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None

t-rekttt, 2022-09-16:
I found out that change **dsrdtr** and/or **rtscts** to false in [this line](https://github.com/babca/python-gsmmodem/blob/ea7db39ade00260ec8fcb045fb8d0d2e683afbd7/gsmmodem/serial_comms.py#L50) will fix it.
Specifically:
dsrdtr = false, rtscts = false: Working
dsrdtr = true, rtscts = false: Working
dsrdtr = false, rtscts = true: Not working

ali-mhz, 2025-03-29:
I have the same problem and above changes didn't work.