babca/python-gsmmodem #105: Modem working but having Timeout Exception

fmaltez, 2022-06-06:
I am able to run the script, and it makes calls and even sends SMS with my E173 modem but I always get exceptions in the code.

What might be the issue (phone number bellow redacted for privacy)?

`
python3 identify-modem.py /dev/ttyUSB0
args: Namespace(baud=115200, debug=False, pin=None, port='/dev/ttyUSB0', wait=0)
Connecting to GSM modem on /dev/ttyUSB0...

== MODEM INFORMATION ==

Manufacturer: huawei
Model: E173
Revision: 11.126.16.00.00

IMEI: 868860007458316
IMSI: 268031602683964

Network: NOS
Signal strength: 16

python3 identify-modem.py -b 115200 -p 0000 -d  /dev/ttyUSB0 
args: Namespace(baud='115200', debug=True, pin='0000', port='/dev/ttyUSB0', wait=0)
Connecting to GSM modem on /dev/ttyUSB0...

== MODEM DEBUG INFORMATION ==

ATI ['Manufacturer: huawei', 'Model: E173', 'Revision: 11.126.16.00.00', 'IMEI: 868860007458316', '+GCAP: +CGSM,+DS,+ES', 'OK']
AT+CGMI: ['huawei', 'OK']
AT+CGMM: ['E173', 'OK']
AT+CGMR: ['11.126.16.00.00', 'OK']
AT+CFUN=?: ['+CFUN: (0-1,4-7),(0-1)', 'OK']
AT+WIND=?: ['COMMAND NOT SUPPORT']
AT+WIND?: ['COMMAND NOT SUPPORT']
AT+CPMS=?: ['+CPMS: ("ME","MT","SM","SR"),("ME","MT","SM","SR"),("ME","MT","SM","SR")', 'OK']
AT+CNMI=?: ['+CNMI: (0,1,2),(0,1,2,3),(0,2),(0,1,2),(0,1)', 'OK']
AT+CVHU=?: ['+CVHU: (0-1)', 'OK']
AT+CSMP?: ['+CSMP: 49,167,0,0', 'OK']
AT+GCAP: ['+GCAP: +CGSM,+DS,+ES', 'OK']
AT+CPIN? ['+CPIN: READY', 'OK']
AT+CLAC: ['&C\r', '&D\r', '&F\r', '&V\r', 'E\r', 'I\r', 'L\r', 'M\r', 'Q\r', 'V\r', 'X\r', 'Z\r', 'T\r', 'P\r', 'D\r', 'A\r', 'H\r', 'O\r', 'S0\r', 'S2\r', 'S3\r', 'S4\r', 'S5\r', 'S6\r', 'S7\r', 'S8\r', 'S9\r', 'S10\r', 'S11\r', 'S30\r', 'S103\r', 'S104\r', '+FCLASS\r', '+ICF\r', '+IFC\r', '+IPR\r', '+GMI\r', '+GMM\r', '+GMR\r', '+GCAP\r', '+GSN\r', '+DR\r', '+DS\r', '+WS46\r', '+CLAC\r', '+CCLK\r', '+CBST\r', '+CRLP\r', '+CV120\r', '+CHSN\r', '+CSSN\r', '+CREG\r', '+CGREG\r', '+CFUN\r', '+GCAP\r', '+CSCS\r', '+CSTA\r', '+CR\r', '+CEER\r', '+CRC\r', '+CMEE\r', '+CGDCONT\r', '+CGDSCONT\r', '+CGTFT\r', '+CGEQREQ\r', '+CGEQMIN\r', '+CGQREQ\r', '+CGQMIN\r', '+CGEQNEG\r', '+CGEREP\r', '+CGPADDR\r', '+CGCLASS\r', '+CGSMS\r', '+CSMS\r', '+CMGF\r', '+CSAS\r', '+CRES\r', '+CSCA\r', '+CSMP\r', '+CSDH\r', '+CSCB\r', '+FDD\r', '+FAR\r', '+FCL\r', '+FIT\r', '+ES\r', '+ESA\r', '+CMOD\r', '+CVHU\r', '+CGDATA\r', '+CSQ\r', '+CBC\r', '+CPAS\r', '+CPIN\r', '+CMEC\r', '+CKPD\r', '+CIND\r', '+CMER\r', '+CGATT\r', '+CGACT\r', '+CGCMOD\r', '+CPBS\r', '+CPBR\r', '+CPBF\r', '+CPBW\r', '+CPMS\r', '+CNMI\r', '+CMGL\r', '+CMGR\r', '+CMGS\r', '+CMSS\r', '+CMGW\r', '+CMGD\r', '+CMGC\r', '+CNMA\r', '+CMMS\r', '+FTS\r', '+FRS\r', '+FTH\r', '+FRH\r', '+FTM\r', '+FRM\r', '+CHUP\r', '+CCFC\r', '+CCUG\r', '+COPS\r', '+CLCK\r', '+CPWD\r', '+CUSD\r', '+CAOC\r', '+CACM\r', '+CAMM\r', '+CPUC\r', '+CCWA\r', '+CHLD\r', '+CIMI\r', '+CGMI\r', '+CGMM\r', '+CGMR\r', '+CGSN\r', '+CNUM\r', '+CSIM\r', '+CRSM\r', '+CCLK\r', '+CLVL\r', '+CMUT\r', '+CLCC\r', '+COPN\r', '+CPOL\r', '+CPLS\r', '+CTZR\r', '+CTZU\r', '+CLAC\r', '+CLIP\r', '+COLP\r', '+CDIP\r', '+CTFR\r', '+CLIR\r', '$QCSIMSTAT\r', '$QCCNMI\r', '$QCCLR\r', '$QCDMG\r', '$QCDMR\r', '$QCDNSP\r', '$QCDNSS\r', '$QCTER\r', '$QCSLOT\r', '$QCPINSTAT\r', '$QCPDPP\r', '$QCPDPLT\r', '$QCPWRDN\r', '$QCDGEN\r', '$BREW\r', '$QCSYSMODE\r', '$QCCTM\r', '^RFSWITCH\r', '^SOFTSWITCH\r', '^FLIGHTMODESAVE\r', '^IMSICHG\r', '^STSF\r', '^STGI\r', '^STGR\r', '^CELLMODE\r', '^SYSINFO\r', '^DIALMODE\r', '^SYSCFG\r', '^SYSCONFIG\r', '^HS\r', '^DTMF\r', '^CPBR\r', '^CPBW\r', '^HWVER\r', '^HVER\r', '^DSFLOWCLR\r', '^DSFLOWQRY\r', '^DSFLOWRPT\r', '^SPN\r', '^PORTSEL\r', '^CPIN\r', '^SN\r', '^EARST\r', '^CARDLOCK\r', '^CARDUNLOCK\r', '^ATRECORD\r', '^CDUR\r', '^BOOT\r', '^FHVER\r', '^CURC\r', '^FREQLOCK\r', '^DSFR\r', '^FREQPREF\r', '^HSPA\r', '^HSUPA\r', '^GPSTYPE\r', '^HSDPA\r', '^GLASTERR\r', '^CARDMODE\r', '^U2DIAG\r', '^RSTRIGGER\r', '^SETPID\r', '^SCSITIMEOUT\r', '^CQI\r', '^GETPORTMODE\r', '^CVOICE\r', '^DDSETEX\r', '^pcmrecord\r', '^CSNR\r', '^CMSR\r', '^CMMT\r', '^CMGI\r', '^RDCUST\r', '^OPWORD\r', '^CPWORD\r', '^DISLOG\r', '^FPLMN\r', '^FPLMNCTRL\r', '^ANQUERY\r', '^RSCPCFG\r', '^ECIOCFG\r', '^IMSICHECK\r', '^USSDMODE\r', '^SLOTCFG\r', '^YJCX\r', '^NDISDUP\r', '^DHCP\r', '^AUTHDATA\r', '^CRPN\r', '^ICCID\r', '^NVMBN\r', '^RXDIV\r', '^DNSP\r', '^DNSS\r', '^WPDST\r', '^WPDOM\r', '^WPDFR\r', '^WPQOS\r', '^WPDSC\r', '^WPDGP\r', '^WPEND\r', '^WNICT\r', '^SOCKETCONT\r', '^WPURL\r', '^WMOLR\r', '^SECTIME\r', '^WPDNP\r', '^WPDDL\r', '^WPDCP\r', 'OK']                         

==dial_callback_demo.py==

python3 dial_callback_demo.py
Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['&C\r', '&D\r', '&F\r', '&V\r', 'E\r', 'I\r', 'L\r', 'M\r', 'Q\r', 'V\r', 'X\r', 'Z\r', 'T\r', 'P\r', 'D\r', 'A\r', 'H\r', 'O\r', 'S0\r', 'S2\r', 'S3\r', 'S4\r', 'S5\r', 'S6\r', 'S7\r', 'S8\r', 'S9\r', 'S10\r', 'S11\r', 'S30\r', 'S103\r', 'S104\r', '+FCLASS\r', '+ICF\r', '+IFC\r', '+IPR\r', '+GMI\r', '+GMM\r', '+GMR\r', '+GCAP\r', '+GSN\r', '+DR\r', '+DS\r', '+WS46\r', '+CLAC\r', '+CCLK\r', '+CBST\r', '+CRLP\r', '+CV120\r', '+CHSN\r', '+CSSN\r', '+CREG\r', '+CGREG\r', '+CFUN\r', '+GCAP\r', '+CSCS\r', '+CSTA\r', '+CR\r', '+CEER\r', '+CRC\r', '+CMEE\r', '+CGDCONT\r', '+CGDSCONT\r', '+CGTFT\r', '+CGEQREQ\r', '+CGEQMIN\r', '+CGQREQ\r', '+CGQMIN\r', '+CGEQNEG\r', '+CGEREP\r', '+CGPADDR\r', '+CGCLASS\r', '+CGSMS\r', '+CSMS\r', '+CMGF\r', '+CSAS\r', '+CRES\r', '+CSCA\r', '+CSMP\r', '+CSDH\r', '+CSCB\r', '+FDD\r', '+FAR\r', '+FCL\r', '+FIT\r', '+ES\r', '+ESA\r', '+CMOD\r', '+CVHU\r', '+CGDATA\r', '+CSQ\r', '+CBC\r', '+CPAS\r', '+CPIN\r', '+CMEC\r', '+CKPD\r', '+CIND\r', '+CMER\r', '+CGATT\r', '+CGACT\r', '+CGCMOD\r', '+CPBS\r', '+CPBR\r', '+CPBF\r', '+CPBW\r', '+CPMS\r', '+CNMI\r', '+CMGL\r', '+CMGR\r', '+CMGS\r', '+CMSS\r', '+CMGW\r', '+CMGD\r', '+CMGC\r', '+CNMA\r', '+CMMS\r', '+FTS\r', '+FRS\r', '+FTH\r', '+FRH\r', '+FTM\r', '+FRM\r', '+CHUP\r', '+CCFC\r', '+CCUG\r', '+COPS\r', '+CLCK\r', '+CPWD\r', '+CUSD\r', '+CAOC\rdial_callback_demo.py', '+CACM\r', '+CAMM\r', '+CPUC\r', '+CCWA\r', '+CHLD\r', '+CIMI\r', '+CGMI\r', '+CGMM\r', '+CGMR\r', '+CGSN\r', '+CNUM\r', '+CSIM\r', '+CRSM\r', '+CCLK\r', '+CLVL\r', '+CMUT\r', '+CLCC\r', '+COPN\r', '+CPOL\r', '+CPLS\r', '+CTZR\r', '+CTZU\r', '+CLAC\r', '+CLIP\r', '+COLP\r', '+CDIP\r', '+CTFR\r', '+CLIR\r', '$QCSIMSTAT\r', '$QCCNMI\r', '$QCCLR\r', '$QCDMG\r', '$QCDMR\r', '$QCDNSP\r', '$QCDNSS\r', '$QCTER\r', '$QCSLOT\r', '$QCPINSTAT\r', '$QCPDPP\r', '$QCPDPLT\r', '$QCPWRDN\r', '$QCDGEN\r', '$BREW\r', '$QCSYSMODE\r', '$QCCTM\r', '^RFSWITCH\r', '^SOFTSWITCH\r', '^FLIGHTMODESAVE\r', '^IMSICHG\r', '^STSF\r', '^STGI\r', '^STGR\r', '^CELLMODE\r', '^SYSINFO\r', '^DIALMODE\r', '^SYSCFG\r', '^SYSCONFIG\r', '^HS\r', '^DTMF\r', '^CPBR\r', '^CPBW\r', '^HWVER\r', '^HVER\r', '^DSFLOWCLR\r', '^DSFLOWQRY\r', '^DSFLOWRPT\r', '^SPN\r', '^PORTSEL\r', '^CPIN\r', '^SN\r', '^EARST\r', '^CARDLOCK\r', '^CARDUNLOCK\r', '^ATRECORD\r', '^CDUR\r', '^BOOT\r', '^FHVER\r', '^CURC\r', '^FREQLOCK\r', '^DSFR\r', '^FREQPREF\r', '^HSPA\r', '^HSUPA\r', '^GPSTYPE\r', '^HSDPA\r', '^GLASTERR\r', '^CARDMODE\r', '^U2DIAG\r', '^RSTRIGGER\r', '^SETPID\r', '^SCSITIMEOUT\r', '^CQI\r', '^GETPORTMODE\r', '^CVOICE\r', '^DDSETEX\r', '^pcmrecord\r', '^CSNR\r', '^CMSR\r', '^CMMT\r', '^CMGI\r', '^RDCUST\r', '^OPWORD\r', '^CPWORD\r', '^DISLOG\r', '^FPLMN\r', '^FPLMNCTRL\r', '^ANQUERY\r', '^RSCPCFG\r', '^ECIOCFG\r', '^IMSICHECK\r', '^USSDMODE\r', '^SLOTCFG\r', '^YJCX\r', '^NDISDUP\r', '^DHCP\r', '^AUTHDATA\r', '^CRPN\r', '^ICCID\r', '^NVMBN\r', '^RXDIV\r', '^DNSP\r', '^DNSS\r', '^WPDST\r', '^WPDOM\r', '^WPDFR\r', '^WPQOS\r', '^WPDSC\r', '^WPDGP\r', '^WPEND\r', '^WNICT\r', '^SOCKETCONT\r', '^WPURL\r', '^WMOLR\r', '^SECTIME\r', '^WPDNP\r', '^WPDDL\r', '^WPDCP\r', 'OK']
DEBUG: write: AT^CVOICE=0
DEBUG: response: ['ERROR']
DEBUG: write: AT^USSDMODE=0
DEBUG: response: ['OK']
INFO: Loading Huawei call state update table
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+35193121314",145', 'OK']
[... 22 lines cut ...]
DEBUG: write: ATD9xxxxxxx2;
DEBUG: response: ['OK']
Traceback (most recent call last):
[... cut ...]
gsmmodem.exceptions.TimeoutException: None

== send_sms_demo.py ==

python3 send_sms_demo.py
Initializing modem...
Sending SMS to: +3519xxxxxxx2
Traceback (most recent call last):
  File "send_sms_demo.py", line 40, in <module>
    main()
  File "send_sms_demo.py", line 30, in main
    response = modem.sendSms(SMS_DESTINATION, SMS_TEXT, True)
  File "/usr/local/lib/python3.8/dist-packages/gsmmodem/modem.py", line 952, in sendSms
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException: None
`

tperalta82, 2022-08-17:
For this specific modem you have, you need to use the notification channel under ttyUSB2, otherwise it won't work

Can close the ticket

ebowwa, 2023-09-17:
may you elaborate? What about with a waveshare sim7600x modem? @tperalta82

tperalta82, 2023-11-12:
> may you elaborate? What about with a waveshare sim7600x modem? @tperalta82

I don't have one of those, so I cannot test nor debug my friend!