babca/python-gsmmodem #106: CMSError: CMS 321

xVinn, 2022-06-23:
```Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1205, in __threadedHandleModemNotification
    self._handleSmsStatusReport(line)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1373, in _handleSmsStatusReport
    report = self.readStoredSms(msgIndex, msgMemory)
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 1428, in readStoredSms
    msgData = self.write('AT+CMGR={0}'.format(index))
  File "build/bdist.linux-armv7l/egg/gsmmodem/modem.py", line 492, in write
    raise CmsError(data, int(errorCode))
CmsError: CMS 321
```
Every time I call for an SMS to be sent using the example code, I get this error after "SMS Delivered"

How do I fix this? Thanks

elite-jwe, 2023-06-05:
Same error here on a Raspberry Pi 4 with a Sim7600X Waveshare Hat : 
`DEBUG: response: ['+CMGS: 40', 'OK']
SMS sent.

DEBUG: write: AT+CMGD=3,0
DEBUG: response: ['OK']
DEBUG: notification: ['+CDSI: "SR",8']
DEBUG: SMS status report received
DEBUG: write: AT+CPMS="SR"
DEBUG: response: ['+CPMS: 0,0,3,23,3,23', 'OK']
DEBUG: write: AT+CMGR=8
DEBUG: response: ['+CMS ERROR: 321']
Exception in thread Thread-11:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/proto/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 1205, in __threadedHandleModemNotification
    self._handleSmsStatusReport(line)
  File "/home/proto/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 1373, in _handleSmsStatusReport
    report = self.readStoredSms(msgIndex, msgMemory)
  File "/home/proto/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 1428, in readStoredSms
    msgData = self.write('AT+CMGR={0}'.format(index))
  File "/home/proto/.local/lib/python3.9/site-packages/gsmmodem/modem.py", line 492, in write
    raise CmsError(data, int(errorCode))
gsmmodem.exceptions.CmsError: CMS 321`

ebowwa, 2023-09-17:
The '+CMS ERROR: 321' is an error code from the GSM modem, which indicates that an invalid memory index was specified in your AT command (AT+CMGR=6). This can happen if you're trying to access a message storage index that doesn't exist or is not accessible. Speaking of which I'm hitting this same issue I plan to check the device's SMS storage configuration and indices. I'll write an update if I when I get better success

ebowwa, 2023-09-17:
try [https://github.com/babca/python-gsmmodem/blob/master/tools/sendsms.py], it should run without a bother use the right flags tho! i.e. /usr/bin/python3 /home/pi/FINAL/python-gsmmodem/examples/send_sms_demo.py -i [PORT] [DESTINATION] "[MESSAGE]"